mne-tools/mne-python #4489: Unable to open EDF+ files from KEMPS

sbang002, 2017-08-14:
I have tried to use mne to open the EDF+ from KEMPS database on MNE but it fails to open.
MNE would be able to recognize that the file is an EDF file but be unable to access it. 
The commands I have run works on other EDF+ files.
I know that the data from KEMPS is not corrupted due to the fact that I was able to open the files perfectly with polymanreport. 
Right now I have tried using the following commands to open the file ... 
         mne.io.read_raw_edf(pth, misc=misc, stim_channel=None, preload=True)
         mne.io.read_raw_edf(pth, stim_channel=auto, preload=True)

teonbrooks, 2017-08-15:
hi @sbang002, do you have a link or a sample of the KEMPS data? that would be helpful for debugging.

sbang002, 2017-08-15:
The following is a data set from the KEMPS data file.
[KempSC.zip](https://github.com/mne-tools/mne-python/files/1225954/KempSC.zip)

teonbrooks, 2017-08-17:
@sbang002, I was able to load the file with 
```python
raw = mne.io.read_raw_edf('KempSC/scorefiles/subid0_visit2-Hypnogram.edf', preload=True)
```
it contains annotation about the different sleep stages. you can access the annotations from `raw._raw_extras[0]['events']`. we don't really have any plotting tool to look at hypnogram data. is there something in particular you wanted to do with the data?

agramfort, 2017-08-19:
@teonbrooks I cannot read these files with current master.

did you try with master?

also these files only contain annotations and no data so we should have a way
to read them directly without using a raw object.

@Slasnista you have worked with such files. do you have a solution? maybe some
code to share?

sbang002, 2017-08-21:
@teonbrooks I tried using the same method as you did to read the files but was unable to succeed. I think that you may have updated code compared to the masters. Currently I get the same problem as stated in my first post even when I use your method.

teonbrooks, 2017-08-21:
I guess something recently happened on master since I last updated it because this is no longer working for me.

@agramfort are you suggesting that we add `read_edf_annotations` function?

agramfort, 2017-08-25:
@sbang002 here is a snippet we wrote with @Slasnista. I guess it can help before we integrate this properly inside MNE

```
import re
import pandas as pd
import mne

def read_edf_annotations(fname):
    """read_edf_annotations

    Parameters:
    -----------
    fname : str
        Path to file.

    Returns:
    --------
    annot : DataFrame
        The annotations
    """
    with open(fname, 'r', encoding='utf-8',
              errors='ignore') as annotions_file:
        tal_str = annotions_file.read()

    exp = '(?P<onset>[+\-]\d+(?:\.\d*)?)' + \
          '(?:\x15(?P<duration>\d+(?:\.\d*)?))?' + \
          '(\x14(?P<description>[^\x00]*))?' + '(?:\x14\x00)'

    annot = [m.groupdict() for m in re.finditer(exp, tal_str)]

    good_annot = pd.DataFrame(annot)
    good_annot = good_annot.query('description != ""').copy()
    good_annot.loc[:, 'duration'] = good_annot['duration'].astype(float)
    good_annot.loc[:, 'onset'] = good_annot['onset'].astype(float)
    return good_annot

def resample_30s(annot):
    annot = annot.set_index('onset')
    annot.index = pd.to_timedelta(annot.index, unit='s')
    annot = annot.resample('30s').ffill()
    annot = annot.reset_index()
    annot['duration'] = 30.
    return annot

annot = read_edf_annotations('./KempSC/subid0_visit1-Hypnogram.edf')
annot = resample_30s(annot)

mne_annot = mne.Annotations(annot.onset, annot.duration, annot.description)

print(mne_annot)
```

sbang002, 2017-08-29:
I think this will work out for me. Thank you for your help

sbang002, 2017-08-29:
The above method worked great for KEMPS but there is another data set called MASS which I thought had the same problem as KEMPS but apparently it doesn't work for MASS. 
The following data set contains data for MASS.
[MASS.zip](https://github.com/mne-tools/mne-python/files/1261511/MASS.zip)

agramfort, 2017-08-29:
@sbang002 here is an example for MASS written by @Slasnista 

this code was written for this work:

https://arxiv.org/abs/1707.03321

please have a look and acknowledge it in your paper (as well as the [MNE papers](http://martinos.org/mne/stable/faq.html#cite)).

We'll be also happy to have a look at your work.

```
import re
import pandas as pd

def read_edf_annotations(fname, annotation_format="edf/edf+"):
    """read_edf_annotations

    Parameters:
    -----------
    fname : str
        Path to file.

    Returns:
    --------
    annot : DataFrame
        The annotations
    """
    with open(fname, 'r', encoding='utf-8',
              errors='ignore') as annotions_file:
        tal_str = annotions_file.read()

    if "edf" in annotation_format:
        if annotation_format == "edf/edf+":
            exp = '(?P<onset>[+\-]\d+(?:\.\d*)?)' + \
                  '(?:\x15(?P<duration>\d+(?:\.\d*)?))?' + \
                  '(\x14(?P<description>[^\x00]*))?' + '(?:\x14\x00)'

        elif annotation_format == "edf++":
            exp = '(?P<onset>[+\-]\d+.\d+)' + \
                  '(?:(?:\x15(?P<duration>\d+.\d+)))' + \
                  '(?:\x14\x00|\x14(?P<description>.*?)\x14\x00)'

        annot = [m.groupdict() for m in re.finditer(exp, tal_str)]
        good_annot = pd.DataFrame(annot)
        good_annot = good_annot.query('description != ""').copy()
        good_annot.loc[:, 'duration'] = good_annot['duration'].astype(float)
        good_annot.loc[:, 'onset'] = good_annot['onset'].astype(float)

    else:
        raise ValueError('Not supported')

    return good_annot

def resample_30s(annot):
    annot = annot.set_index('onset')
    annot.index = pd.to_timedelta(annot.index, unit='s')
    annot = annot.resample('30s').ffill()
    annot = annot.reset_index()
    annot['duration'] = 30.
    return annot

if __name__ == "__main__":

    # read mass with format edf++
    annot_mass = read_edf_annotations(
        './MASS/subjectid1.edf',
        annotation_format="edf++")
```

Slasnista, 2017-08-29:
@sbang002, indeed MASS is encoded in edf++ which is slightly different from traditional edf/edf+ encoding. For more details, you can have a look at http://51.15.173.162/en/contribute/edf/

Please note that there might be annotations different from sleep stages in the returned `good_annot`, especially for MASS dataset. When using the code above, you might want to get rid of them in order to work only with sleep stages.

sbang002, 2017-09-05:
Is it possible for met to attain the orig_time  attribute from the EDF+ and EDF++? 
I thought I would be able to pull it out since mne.Annotations has this attribute but I was unsuccessful in doing so.  I think something must be added into the parsing code to get this attribute but I don't know how I would attain it.

Slasnista, 2017-09-05:
Hi @sbang002, you shall get this information through the `raw` object constructed by `mne.io.read_raw_edf()`. More precisely `raw.info["meas_date"]` should give you the start date / time of the record that you can convert then into YY/MM/DD - hh/mm/ss - ...

sbang002, 2017-09-06:
@Slasnista  I cannot use mne.io.read_raw_edf() because of the same problem as this issue.

agramfort, 2017-09-06:
maybe you give a try to a fix based on the code we gave you?

bdyetton, 2018-09-11:
@agramfort @sbang002  was an intern working in our lab. The code posted here worked for us. Please feel free to close this issue and apologies for leaving it open so long.

agramfort, 2018-09-11:
do we have the problematic files? is it still broken?

cc @Slasnista

bdyetton, 2018-09-16:
Yes, this code above coupled with:

```
    try: #regular
        EDF_file = mne.io.read_raw_edf(path, stim_channel='auto', preload=True, verbose=False)
        raw_annot = mne.io.find_edf_events(EDF_file)
        annot = pd.DataFrame(raw_annot, columns=['onset', 'duration', 'description'])
    except TypeError: #edf+
        annot = read_edf_annotations(path, annotation_format="edf/edf+")
    except ValueError: #edf++
        annot = read_edf_annotations(path, annotation_format="edf++")
```
works well, thanks!

bdyetton, 2019-07-18:
This issue can be closed if you like.

agramfort, 2019-07-18:
does it work with the new 

`mne.read_annotations`

?

bdyetton, 2019-07-18:
Yep, works well, thanks.

closed 2019-07-22 by massich