mne-tools/mne-python #4497: Rename and relocate `get_edf_events`

teonbrooks, 2017-08-17:
I just noticed that we now have a new function `mne.io.get_edf_events`, which is used to extract events from the EDF family. I was thinking two things:  
1. shouldn't this belong at the top level mne. The logic behind this is that we have the function `mne.find_events` which exists there. 
2. this should be renamed `find_edf_events` to follow the naming pattern.

agramfort, 2017-08-17:
edf events are more like annotations with onset, duration + description. I
think it's a mistake to call them events which are different for MNE/Epochs
etc.

larsoner, 2017-08-17:
This was introduced this cycle so if we're going to change the name we should do it ASAP to avoid needing to deprecate

cbrnr, 2017-09-19:
Unfortunately, EDF/BDF files can have events encoded either (or both) as a stim channel or in an event table. Whereas `mne.find_events` works only on the stim channel, `mne.io.get_edf_events` relies only on the event table. I'm not sure if renaming it to `find_edf_events` is a good idea, because people might get confused as to which function to use for EDF files. Granted, people could already be confused by these two functions right now, but `mne.find_events` works for all kinds of file types whereas `mne.io.get_edf_events` only works on the EDF family, which is why it should probably stay in the `io` package.

agramfort, 2017-09-19:
I think find_edf_events naming is more consistent

I agree it should stay in the io module

cbrnr, 2017-09-20:
@jaeilepp?

agramfort, 2017-09-20:
@jaeilepp <https://github.com/jaeilepp> is gone and back home in Finland
... he left my group :'(

cbrnr, 2017-09-20:
Oh, I didn't know that - he added this new method recently. I can create a PR and rename it (but leave it in `io`) if you want.

agramfort, 2017-09-20:
please do

PR #4588 (2017-09-21): Rename get_edf_events to find_edf_events
Fix #4497
closed 2017-09-21 by agramfort