mne-tools/mne-python #4499: DOC: Return value of ICA.apply

aweinstein, 2017-08-17:
The method ICA.apply returns the processed object. However, its [documentation](https://martinos.org/mne/stable/generated/mne.preprocessing.ICA.html#mne.preprocessing.ICA.apply) does not list the return value.

Along the same lines, ICA.apply seems to change in place the data instance. It might be a good idea to mention this behavior in the documentation (as a new user I was surprised by this).

agramfort, 2017-08-17:
indeed

can you send a PR to update the docstring?

thanks

aweinstein, 2017-08-17:
I can, and I will. But before doing that, can you clarify the rationale for changing the raw data in place, and also returning the modified data? 

I ask because perhaps is better to keep only one of the two behaviors.

Alejandro

larsoner, 2017-08-17:
It goes against what most Python packages do, but we decided it was worth it to have long chains like:
```
raw.copy().load_data().pick_types(meg=True).filter(None, 40)
```

aweinstein, 2017-08-17:
I just send PR #4500.

closed 2017-08-18 by larsoner