mne-tools/mne-python #4502: API: picks as list of string

kingjr, 2017-08-18:
Picking a particular channel is currently relatively complex for new users.

It either involves pythonic tricks:
```python
picks = [ii for ii, ch in enumerate(epochs.ch_names) if ch == 'mychan']
epochs.plot(picks=picks)
```

Or master of MNE's functions:
```python
picks = mne.pick_channels(evoked.info. [`my_chan`])
epochs.plot(picks=picks)
```

Since picks is always referring to a list of `int` and `ch_names` is always a list of `str`, I propose that we accept picks ot be a list of `str`:
```python
epochs.plot(picks=['my_chan'])
```

dengemann, 2017-08-18:
@kingjr I had thought about this for long time and am glad you bring it up.
I think it can be a good idea. It will be unproblematic whenever the
function picks are passed to has access to .info.
I sometimes thought we should have written mne.pick_types that way to
accept a list of channel keys. But I would not suggest to depracate it.
Maybe a keys argument could be added there.

kingjr, 2017-08-18:
@dengemann For pick_types we could accept `picks` to be `dict`

dengemann, 2017-08-18:
I don't yet see what you have in mind. The function takes keyword arguments
for channel types. It is annoying that you cannot just loop over types but
have to. Create bool flags and then deal with meg that can be 'mag', 'grad'
or True/False.
If you could do:

pick_types(info, keys=('mag', 'eeg', 'eog'))

That would simplify lots of code.

kingjr, 2017-08-18:
Sorry, I meant accepting the parameter `picks` to be a list of `int` (for indexing), a list of `str` (to find channel names) or a `dict` (to quickly select channel types).
```python
evoked.plot(picks=dict(grad=True, eeg=False))
evoked.plot(picks=range(10))
evoked.plot(picks=['MEG 1132', 'MEG 1022'])
```

dengemann, 2017-08-18:
Yes. I think we are discussing complementary ideas. I like the list of
types notation more as it is far more concise. It could not only simplify
passing picks but creating them. In some situations you would still need
the full integer arrays.

kingjr, 2017-08-18:
I see. However, the issue is that a list of types would conflict with a list of names.
```python
evoked.plot(picks=['MEG 1132', 'MEG 1022'])
evoked.plot(picks=['mag', 'grad'])
```
I'm thus in favor of **Proposal 2**, i.e. accepting list of string and dict. 

I'll update the above description.

larsoner, 2017-08-18:
@kingjr I updated your proposal to to be Py3k-friendly, and to reorder with `list(range(...))` first (since it's what we have now)

larsoner, 2017-08-18:
It does seem like proposal 2 would just require a ~6-line private function anytime we use `picks`, so it's not so bad from an implementation standpoint. And from a user standpoint it's pretty simple.

It does have overlap, e.g. `inst.pick_types(['EEG 001', 'EEG 002'])` will be the same as `inst.pick_channels(['EEG 001', 'EEG 002'])`, but I can live with that.

@kingjr if your sense is that this will reduce pain substantially for new users, +1 from me

dengemann, 2017-08-18:
I would still prefer proposal 1 (list of channel types). It would allow me
to kill lots of code in virtually any project. I don't think we would ever
hit a case where channel names and type names are identical. For updating
mne.pick_types along these lines we would anyways have to add a new keyword.

kingjr, 2017-08-18:
> I would still prefer proposal 1 (list of channel types). It would allow me to kill lots of code in virtually any project. 

Can you describe in what way Proposal 2 wouldn't?

> I don't think we would ever hit a case where channel names and type names are identical. 

This would still require a "smart" guess; not safe IMO. 

> For updating mne.pick_types along these lines we would anyways have to add a new keyword.

Why? We wouldn't need to update `pick_types` in any of these proposal

larsoner, 2017-08-18:
> Why? We wouldn't need to update pick_types in any of these proposal

Agreed, we would just update any function that has `picks` argument. Instead of taking only list of int, take the other things, too. `pick_types` and `pick_channels` would remain the same.

dengemann, 2017-08-18:
I would independently from these proposals like to get rid of the dict /
pick_types api because it is too irregular and makes looping over channel
types a pain.
In terms of elegance ('mag', 'eeg') is much more convenient then
dict(meg='mag', eeg=True).

kingjr, 2017-08-18:
I agree; although for this we can propose an orthogonal PR where mne.pick_types accepts a list of str
```python
mne.pick_types(info, types=('mag', 'grad', 'eog'))
```

dengemann, 2017-08-18:
Sounds like a good plan. Once you have picks you can use it downstream the
old way.

choldgraf, 2017-08-18:
I'm +1 with 

`mne.pick_types(info, types=('mag', 'grad', 'eog'))`

I think this is much more descriptive than having a gigantic list of true/false.

I'm also +1 for allowing lists of strings. I'm a little hesitant to allow for `dict(eeg=True)` since I agree that is a clunky API already.

What if instead of `dict(grad=True..`, we allowed `dict(types=['eeg', 'meg', ...])`?

larsoner, 2017-08-18:
I think that `picks=dict(types=['eeg'])` is about the same clunkiness as `picks=dict(eeg=True)`.

Thinking about it more I actually don't mind `picks=['eeg', 'meg']`. I think that collisions will be *extremely* rare. In those cases where they do collide, we can raise an error about ambiguity and tell people to use `picks=pick_channels` or `picks=pick_types` explicitly.

choldgraf, 2017-08-18:
I'm +1 with that. Just do something like

```
PICK_TYPES = ['eeg', 'meg', 'ecog', 'blahblah']`
if any(any(ii == np.array(data.ch_names)) for ii in PICK_TYPES):
    warn('wtf dude')
```

larsoner, 2017-08-18:
It will be slightly more complicated because we need to check if all are channels, or all are types, and raise an error if it doesn't fit exactly one of those criteria, but yes something like that.

kingjr, 2017-08-18:
ok, I'm updating the proposal sheet, wait for what @agramfort think to make any attempt PR, as API change isn't the least controversial to do.

larsoner, 2017-08-18:
We just need to ensure he first sees that #3842 was merged so that he's in a good mood when reading this proposal

choldgraf, 2017-08-18:
I have another proposal: what if we allow users to supply a dataframe to the `picks` argument. This dataframe would contain a single column, `'data'`, and each row of this column would be another dataframe that would contain a list of `Series` objects, one for each data type to be plotted.

choldgraf, 2017-08-18:
also we can deprecate the `picks` argument and make it `pick_pandas`

kingjr, 2017-08-18:
@choldgraf I actually thought of this, it'd be make so much more sense.... accessing the channel position, type etc would be so simple...

larsoner, 2017-08-18:
Only if we can do a dataframe of dataframes, each with a single column for each channel type

Just so we're clear, my vote is for Proposal 4. I'm going to update the proposal to put my +1 there, everyone else feel free to do the same (people without permission to change, leave a comment and we'll update)

larsoner, 2017-08-18:
@nbara you had an overall :+1: feel free to have another look at the top comment if you have time

choldgraf, 2017-08-18:
so simple!

[image]

(in seriousness, I'm +1 with proposal 4 too)

kingjr, 2017-08-18:
> Only if we can do a dataframe of dataframes, each with a single column for each channel type

?
```python
epochs.info['chs'] = pd.DataFrame(dict(name='MEG 001', type='mag', pos=xyz, ..., ...))
# find chan type
epochs.ch_names.query('type=="mag"')
# find specific combinations
epochs.ch_names['name'].apply(lambda x: 'meg 10' in x)
```

choldgraf, 2017-08-18:
@kingjr you aren't supposed to expose our secret plot to slowly introduce pandas into everything in MNE!

larsoner, 2017-08-18:
@kingjr that was a joke, using Pandas for this sort of thing seems way overkill

dengemann, 2017-08-18:
Guys let's do one revolution at a time. Overcoming the picks dict would
already be a great step for mnekind.

dengemann, 2017-08-18:
Thinking of mne.pick_types I think we never had a single example supporting
meg/eeg/eog/ etc. as positional arguments.

We could indeed dare to make types the 2nd argument:

picks = mne.pick_types(raw.info, ('mag', 'eeg'))

That would be simply beautiful.

choldgraf, 2017-08-18:
IMO that is the most conceptually simple way to pick types, +1

dengemann, 2017-08-18:
That wouldn't even necessitate deprecation of anything I would say.
@choldgraf @kingjr @larsonser

nbara, 2017-08-18:
I'm +1 for Proposal 4 as well!

agramfort, 2017-08-19:
ok I coming late here but my first reaction is that "There should be one-- and preferably only one --obvious way to do it."

regarding your proposal 4 that seems to satisfy most of you:

```
evoked.plot(picks=list(range(10)))  -> already works
```

```
evoked.plot(picks=['MEG 1132', 'MEG 1022'])  # can be done with:
evoked.copy().pick_channels(['MEG 1132', 'MEG 1022']).plot()
```

```
evoked.plot(picks=dict(grad=True, eeg=False))  # can be done with:
evoked.copy().pick_types(meg='grad', eeg=False).plot()
```

by putting the picking in the data container we don't need to have a complex pick logic in every viz or computation function.

thoughts?

larsoner, 2017-08-19:
ok I coming late here but my first reaction is that "There should be one--
and preferably only one --obvious way to do it."

I agree as a general principle, and this does clearly violate it. The
question in my mind is: does the added simplicity and meeting of users
expectations justify it?

> by putting the picking in the data container we don't need to have a
complex pick logic in every viz or computation function.

It's still a bit harder. Most users probably want picks to do what 4
proposes, and have some disappointment when it doesn't.

Well the complex logic wouldn't exactly be in every function. It would be
in a single private function (that gets called by each function). But
conceptually this replaces one DRY function that should already exist but
probably doesn't (to validate picks are list-like of int and are all valid
values).

dengemann, 2017-08-19:
@agramfort think of how our proposal would help looping over channel types
to get picks. At least for pick_types. You just loop over ('meg', 'eeg')
etc.

The dict of bool / string demanded by its current API is mildly spoken a
PITA. You need to make exceptions for "meg" which can be
True/False/"grad"/"mag". Then it is not parsimonious. Why eeg=True if it
can just be "eeg".
If we could overcome for once our purist API change culture we would make
certainly 99% of all users significantly more happy.
Or differently put the one way to do it that we currently suggest is not
the right / best one.

Please think of it twice.

agramfort, 2017-08-20:
how would you write this:

```
picks = pick_types(raw.info, meg=False, eeg=True, stim=False, eog=False,
                   exclude='bads')

epochs = Epochs(raw, events, event_id, tmin, tmax, proj=False,
                picks=picks, baseline=None, preload=True,
                verbose=False)
```

would we deprecate some pick_ methods or functions to avoid duplications?

choldgraf, 2017-08-20:
wouldn't you just do:

`Epochs(raw, events, event_id, tmin, tmax, ..., picks=['eeg'], ...`

or if you wanted a subset of channels:

`Epochs(raw, events, event_id, tmin, tmax, ..., picks=['ch_01', 'ch_03'])`
?

agramfort, 2017-08-20:
this would exclude bads automatically?
don't you make a difference between ('eeg') and ['eeg'] ie tuple for type
and list of ch names?

choldgraf, 2017-08-20:
ah I see what you mean - in that case I think it'd be fine to include an `exclude` kwarg people can use in `Epochs`...it seems like a natural thing to do.

re: type vs. list of names, I think @larsoner's thought was that it would be rare enough not to be a problem and that we could handle it with good warnings or errors. That said, I agree it feels clunky

larsoner, 2017-08-20:
In practice I don't expect collisions, and when they do happen it's easy to
redirect to functions.

I'd say behave as exclude='bads' and if you want something more advanced
(e.g., bads or ignore_ref non-defaults) use the functions. -1 on adding
extra kwargs to deal with all of these cases.

In my mind this is about improving convenience and removing barriers to
entry for minimal maintenance and implementation cost, not replacing the
existing, more complete API.

nbara, 2017-08-21:
@agramfort If I may, as a relatively new user, I find channel picking in MNE to be extremely unintuitive. Syntaxes like these in particular seem very convoluted to me :

```python
evoked.copy().pick_channels(['MEG 1132', 'MEG 1022']).plot()
evoked.copy().pick_types(meg='grad', eeg=False).plot()
```

I appreciate that an API change is not insignificant, but in this case I would find it most welcome. 

(just my 2¢)

agramfort, 2017-08-21:
@nbara would you suggest to also simplify these lines:

evoked.copy().pick_channels(['MEG 1132', 'MEG 1022'])
evoked.copy().pick_types(meg='grad', eeg=False)

ie without the plot? or it's the fact that you need to pick data before
plotting?

dengemann, 2017-08-21:
@agramfort 

```Python
picks = pick_types(raw.info, types=('eeg',), exclude='bads')

epochs = Epochs(raw, events, event_id, tmin, tmax, proj=False,
                picks=picks, baseline=None, preload=True,
                verbose=False)

```

We would not need to deprecate anything.
Once types is not None, all other flags will be ignored.
Already updating this function would clean up the code as it allows you to get picks very easily:

```Python

for ch_type in ('mag', 'grad', 'meg', 'eeg'):
    picks = mne.pick_types(raw.info, types=(ch_type,))
    # or
    picks = mne.pick_types(raw.info, (ch_type,))
    pass
```

agramfort, 2017-08-21:
What I am thinking is that we could have a single pick method

evoked.pick(picks, channels, types)

so what @nbara suggests would be written:

evoked.copy().pick(channels=['MEG 1132', 'MEG 1022']).plot()
evoked.copy().pick(types=['grad']).plot()

so the semantic of picks stays picks=list of int

dengemann, 2017-08-21:
Which would be compatible with the upgrade of mne.pick_types which always
returns array of int. Keep in mind that we need picks in data that don't
have info, pure numpy.

agramfort, 2017-08-21:
would we still need to refactor pick_types?

you envision a pb with Covariance class?

dengemann, 2017-08-21:
I need mne.pick_types all the time. But taking what you and @nbara suggest
to the next level we could have:

mne.pick(info, types, channels=None, exclude='bads', include=None)

nbara, 2017-08-21:
> What I am thinking is that we could have a single pick method

Yes that would be good IMO !

> or it's the fact that you need to pick data before plotting?

Yes this is confusing too (but maybe another issue)

@dengemann's suggestion is definitely an improvement in my view

larsoner, 2017-08-21:
I don't like these proposals as much as the existing number 4.

The unified pick function is ambiguous / less obvious to me. For example,
what happens when I pass types and names? Is it and-like or or-like?

For the unified pick as an alternative (even though from an API perspective
it's orthogonal) to the picks string list, functions that take a picks
argument (like Epochs) are still overly complex for most use cases. People
will still reasonably wish that they could directly pass a list of string
(as channels or types), and we will continue to disappoint them by not
allowing it.

The copy-pick-operate proposal can be (very) memory intensive compared to
the picks-argument equivalent, too. This can especially be three case in
the common raw-filter-epoch loop.

It's also not the same -- something like raw.filter can only reasonably be
done with picks-argument way. So pursuing copy-pick-operate as the ideal
pattern is a bit of a non-starter from my end.

agramfort, 2017-08-21:
hum... the world is against me. Yet:

"il y a que les imbéciles qui changent pas d'avis"

so every function/method that has picks as parameter now will first call

picks = check_picks(info, picks)

where check_picks would be in utils.py ?

@larsoner if we do this we should do it without increasing test time.

jasmainak, 2017-08-21:
I honestly find the existing `picks` API quite intuitive and well designed, but what I find missing is a clear documentation or tutorial.

For me, `picks` is nothing but a list of indices and all the `pick_xxx` functions are helpers to compute these indices. If you are using a scripting language like Matlab or Python, indexing is one of the first thing you learn. 

If you change the semantics of what `picks` is (e.g., in Proposal 4), this will require a change in the code not just in code but also in other repos (like autoreject). [@dengemann up for another PR? :P ]

Please consider carefully the implications of this change

nbara, 2017-08-21:
@jasmainak Not sure I agree with you. 

Like many others I'm guessing, I spend most of my analysis time doing exploratory work, i.e. quickly looking at my data to get a "feel" of how they look, or check that e.g. an basic ERP is here or not. 

And to do this the current system is particularly unintuitive. For instance I'll want to look at channel Oz, or Cz, and I don't care what index they correspond to (esp. given that these indices might change depending on the layout), ideally I'd just want to do `evoked.plot('Cz')`, and then quickly see I I get something similar in another channel `evoked.plot('Cz')`. Then I might wan to try new filter settings on all my channels before plotting again. This takes quite a lot of keystrokes currently. Not to mention that the process of copy-picking before plotting is just... weird.

agramfort, 2017-08-21:
what I realize with this conversation is that while EEG users know the name
of their channels MEG people don't.... they are just stupid names and
numbers.

@nbara do you know this PR:
https://github.com/mne-tools/mne-python/pull/4221

choldgraf, 2017-08-21:
*and ECoG people have 0 ability to predict their channel names ahead of time

jasmainak, 2017-08-21:
@nbara have you used [`plot_topo`](https://martinos.org/mne/stable/generated/mne.viz.plot_evoked_topo.html#mne.viz.plot_evoked_topo)? It seems to be exactly for your requirement.

jasmainak, 2017-08-21:
> No, proposal 4 adds functionality, it does not remove existing functionality of None or list of int, i.e. it is backward compatible, unless you have some code that relies on errors being thrown for non-int lists.

I think we would have to handle non-int lists as you correctly anticipate. When it is `int`, it's really easy to index matrices without thinking twice. You're probably proposing that `picks` is internally handled still as an `int` but I guess that would need additional code ...

nbara, 2017-08-21:
@jasmainak No I wasn't aware of plot_topo, thanks ! Does it work for TF data?
@agramfort #4221 looks very neat indeed!

>MEG people don't.... they are just stupid names andnumbers.

O_O I used to work with MEG and I knew exactly where my channels were (on a CTF system, I used to plot a bunch of channels together using wildcards such as MRT* to get all right temporal sensors etc.).  
As for ECoG, well yes... But wouldn't you just have some kind of map/image showing electrode implantation and use that to look at your activations for each patient ? I mean, it's not about knowing the location in advance, in some cases it's just more convenient to plot stuff by some kind of label instead of by number. Or maybe that's just me :) 

Anyway, I don't mean to pollute this thread too much. Still in favour of Proposal 4.

jasmainak, 2017-08-21:
> @jasmainak No I wasn't aware of plot_topo, thanks ! Does it work for TF data?

yep, take a look [here](http://martinos.org/mne/stable/generated/mne.time_frequency.AverageTFR.html#mne.time_frequency.AverageTFR.plot_topo) :)

[3 more replies]

PR #4511 (2019-02-13): MRG: Allow str list in picks
~~WIP as proof of concept including changes to `mne/viz` functions (and a few other misc ones) as requested by @agramfort.~~

Assuming people are convinced, need to do:

* [x] complete tests for `_picks_to_idx`
* [x] update the introduction (or other?) tutorial
* [x] properly document options
* [x] go through codebase and unify all `picks` logic
* [x] make CIs happy
* [x] finalize docstring via feedback
* [x] replace all XXX with finalized docstring, or use `doccer` depending on #5753

Closes #4502.
closed 2019-02-13 by larsoner