mne-tools/mne-python #4507: Doubts about pipeline and ICA

DominiqueMakowski, 2017-08-21:
Dear MNErs, I've got several questions regarding the processing pipeline I'm trying to set up. 

First of all, here's what I do for each participants (using mne 15) : 

```python
import neurokit as nk
import mne
import autoreject

...
# Reading raw and loading conditions list
...

print("    - Filtering")
raw.filter(1, 40, picks=None, filter_length='auto', l_trans_bandwidth='auto', h_trans_bandwidth='auto', n_jobs=1, method='fir', iir_params=None, phase='zero', fir_window='hamming', verbose=None, fir_design="firwin")

print("    - Epoching")
raw, events, event_id = nk.eeg_add_events(raw, events_channel="PHOTO", cut="lower", conditions=conditions)
epochs = mne.Epochs(raw, events, event_id, tmin=-0.25, tmax=3, baseline=(None, 0), reject=None, verbose=False, detrend=1, preload=True)
epochs.resample(200, npad="auto")

print("    - ICA")
reject = autoreject.compute_thresholds(epochs, picks=mne.pick_types(epochs.info, meg=False, eeg=True), verbose=False)  
ica = mne.preprocessing.ICA(n_components=0.95, method='fastica', max_iter=500)
ica = ica.fit(epochs, picks=mne.pick_types(epochs.info, meg=False, eeg=True), decim=3, reject=reject)
# Note that this, for several participants, returns a RuntimeError: one PCA component captures most of the variance, please increase treshold. But it does that even if I set n_components to 0.999 :/
ica = ica.detect_artifacts(epochs, eog_ch=epochs.copy().pick_types( meg=False, eeg=False, eog=True).ch_names, eog_criterion=0.1, skew_criterion=-0.5, kurt_criterion=-0.5, var_criterion=0)
epochs = ica.apply(epochs)

print("    - Autoreject")
ar =  autoreject.LocalAutoRejectCV(verbose=False, picks=mne.pick_types(epochs.info, meg=False, eeg=True))
epochs = ar.fit_transform(epochs)

print("    - Rereference")
epochs.set_eeg_reference(ref_channels=None, projection=False)
```

My goal is to do ERP analysis mainly, and especially looking at the late positive potentials (LPP) elicited by emotional pictures.

My questions are:
1) Where is the best place to rereference the signal? I did it at the end but maybe it's better at the beginning?
2) "average" reference *seems* to look somewhat better than ["TP9", "TP10"] (the reference that people told me to do). Is this normal? Are there any plans about implementing other types of referencing such as REST (See [this](http://www.sciencedirect.com/science/article/pii/S1388245710004153) or [this](http://iopscience.iop.org/article/10.1088/0967-3334/22/4/305/meta) paper)? 
3) Is resampling done at this stage ok?
4) ICA *seems* to severely alter the signal, adding noise: Here's the plot, over a susbset of central electrodes, of the grand averages by conditions with ICA (picture above) and with the ICA block commented off (picture below).

[image]
[image]

5) Finally, do you have any general advices about how to enhance that pipeline? 

Thanks **a lot** for any comments and suggestions!

jasmainak, 2017-08-21:
> Note that this, for several participants, returns a RuntimeError: one PCA component captures most of the variance, please increase treshold. But it does that even if I set n_components to 0.999 :/

You can avoid this problem by setting `n_components` to `None`.

Also I have never used `ica.detect_artifacts()`. Did you check what components were detected? I have suspicions that this may be the source of your problems ...

DominiqueMakowski, 2017-08-23:
I've opted for the more classic approach:

```python
reject = autoreject.compute_thresholds(epochs, picks=mne.pick_types(epochs.info, meg=False, eeg=True), verbose=False)
ica = mne.preprocessing.ICA(n_components=None, method='fastica', max_iter=500).fit(epochs)
# EOG
eog_epochs = mne.preprocessing.create_eog_epochs(raw)
eog_inds, scores = ica.find_bads_eog(eog_epochs)
eog_inds = eog_inds[:2]
ica.exclude += eog_inds
epochs = ica.apply(epochs)
```
And it does not alter the signal like the previous method. Also, setting n_components to None did work. Thanks!

jona-sassenhagen, 2017-08-30:
Can this be closed?

dengemann, 2017-08-30:
Btw. Dominque, for purely usage related questions / support please use the
mailing list. Github is for development and maintenance of functionality.

DominiqueMakowski, 2017-08-30:
@dengemann Noted!
@jona-sassenhagen yep

closed 2017-08-30 by DominiqueMakowski