mne-tools/mne-python #4513: incorrect channel locations in mne/channels/data/montages/10-5_EGI129.csd

sidneycollin, 2017-08-23:
When comparing a 2D plot of the 129 EGI coordinates given in  [mne/channels/data/montages/10-5_EGI129.csd](https://github.com/mne-tools/mne-python/blob/916b168270d01d81f3cc6286190aa83bdceee6bd/mne/channels/data/montages/10-5_EGI129.csd) to the 2D plot of the coordinates supplied by the EGI company, it seems that some electrode locations are off. I attached a picture of the two plots to this issue. Channel 15 is highlighted to show an example of one of the electrodes whose position is off. In the official EGI channel map, channel 15 is on the center line with channels 16, 11, 6, etc., whereas in the 2D plot of the locations given in the mne file, channel 15 if off center. Can somebody shed light onto why this is the case, or how to fix it? 

[image]

Here is the code i used to plot the montage: 
`import pandas as pd
import matplotlib.pyplot as plt

import mne

channel_names = [str(i+1) for i in range(129)]
montage = mne.channels.read_montage(kind='10-5_EGI129', ch_names=channel_names)

pos_2d = montage.get_pos2d()

fig = plt.figure(1)
ax = fig.add_subplot(111)

for i, vals in enumerate(pos_2d):
    ax.scatter(vals[0], vals[1], color='b') 
    ax.text(vals[0], vals[1], str(i+1), size=10, zorder=1,  
    color='k') 

ax.set_xlabel('x')
ax.set_ylabel('y')
plt.show()`

Thank you!

agramfort, 2017-08-23:
are you sure these positions are available in the csd file?

To make your own montage form this csd file you can use:

import os
import numpy as np
import pandas as pd
import matplotlib.pyplot as plt

import mne
from mne.channels import Montage

fname = os.path.join(os.path.dirname(mne.__file__),
                     'channels/data/montages/10-5_EGI129.csd')

data = np.genfromtxt(fname, dtype='str', skip_header=2)
ch_names = data[:, 0].tolist()
pos = data[:, 4:7].astype(float)
selection = np.arange(len(pos))

montage = Montage(pos=pos, ch_names=ch_names, kind="10-5_EGI129",
                  selection=selection)

dengemann, 2017-08-23:
This is not the right montage. The csd one is not anylonger supported and
was removed recently.

The regular EGI montages should be these ones:

mne/channels/data/montages/GSN-HydroCel-128.sfp

mne/channels/data/montages/GSN-HydroCel-129.sfp

mne/channels/data/montages/GSN-HydroCel-256.sfp

mne/channels/data/montages/GSN-HydroCel-257.sfp

mne/channels/data/montages/GSN-HydroCel-32.sfp

mne/channels/data/montages/GSN-HydroCel-64_1.0.sfp

mne/channels/data/montages/GSN-HydroCel-65_1.0.sfp

sidneycollin, 2017-08-23:
I know it's not an issue with the montage because if I read the file without the read_montage() function, I still see the same issue. Here is the code I was using previously (without the read_montage() function):
```
import os

import mne
import matplotlib.pyplot as plt
import pandas as pd

electrode_num = 129

filename = os.path.join(os.path.dirname(mne.channels.__file__), 'data', 'montages', '10-5_EGI129.csd')
df = pd.read_csv(filename, sep='\s{2,}', header=1)
df.rename(columns = {'// Label':'Label'}, inplace = True)
df = df[~df['Label'].str.contains("[a-zA-Z]").fillna(False)]

df.index = df['Label']

fig = plt.figure(1)
ax = fig.add_subplot(111)

for i, row in df[['X','Y']].iterrows(): # plot each point + its index as text above
    ax.scatter(row.X, row.Y, color='b')
    ax.text(row.X, row.Y,  str(i), size=10, zorder=1, color='k')

ax.set_xlabel('x')
ax.set_ylabel('y')
plt.show()

```
Here is a picture of the plot that you get after running this code:
(Note that channel 15 is still off center.)
[image]

sidneycollin, 2017-08-24:
ok, thank you!

closed 2017-08-24 by sidneycollin