mne-tools/mne-python #4518: Can't run unittests, since '-a' is not recognized as a n option by my version of pytest

smgutstein, 2017-08-28:
When I try to run unittests, I get the following error:

```
$ make test
python setup.py build_ext -i
running build_ext
rm -f .coverage
py.test -a 'not ultra_slow_test' mne
usage: py.test [options] [file_or_dir] [file_or_dir] [...]
py.test: error: unrecognized arguments: -a
  inifile: None
  rootdir: ~/Downloads/mne-python
Makefile:41: recipe for target 'test' failed
make: *** [test] Error 2
```
The version of py.test (which is seemingly the same as pytest !?) I have installed is:

```
$ py.test --version
This is pytest version 3.0.7, imported from /usr/local/lib/python2.7/dist-packages/pytest.pyc
setuptools registered plugins:
  pytest-xdist-1.20.0 at /usr/local/lib/python2.7/dist-packages/xdist/looponfail.py
  pytest-xdist-1.20.0 at /usr/local/lib/python2.7/dist-packages/xdist/plugin.py
  pytest-forked-0.2 at /usr/local/lib/python2.7/dist-packages/pytest_forked/__init__.pyc
  pytest-cov-2.5.1 at /usr/local/lib/python2.7/dist-packages/pytest_cov/plugin.py

```

When I run it with --help, I get this:

```
$ py.test --help
usage: py.test [options] [file_or_dir] [file_or_dir] [...]

positional arguments:
  file_or_dir

general:
  -k EXPRESSION         only run tests which match the given substring
                        expression. An expression is a python evaluatable
                        expression where all names are substring-matched
                        against test names and their parent classes. Example:
                        -k 'test_method or test_other' matches all test
                        functions and classes whose name contains
                        'test_method' or 'test_other'. Additionally keywords
                        are matched to classes and functions containing extra
                        names in their 'extra_keyword_matches' set, as well as
                        functions which have names assigned directly to them.
  -m MARKEXPR           only run tests matching given mark expression.
                        example: -m 'mark1 and not mark2'.
  --markers             show markers (builtin, plugin and per-project ones).
  -x, --exitfirst       exit instantly on first error or failed test.
  --maxfail=num         exit after first num failures or errors.
  --strict              run pytest in strict mode, warnings become errors.
  -c file               load configuration from `file` instead of trying to
                        locate one of the implicit configuration files.
  --continue-on-collection-errors
                        Force test execution even if collection errors occur.
  --fixtures, --funcargs
                        show available fixtures, sorted by plugin appearance
  --fixtures-per-test   show fixtures per test
  --import-mode={prepend,append}
                        prepend/append to sys.path when importing test
                        modules, default is to prepend.
  --pdb                 start the interactive Python debugger on errors.
  --pdbcls=modulename:classname
                        start a custom interactive Python debugger on errors.
                        For example:
                        --pdbcls=IPython.terminal.debugger:TerminalPdb
  --capture=method      per-test capturing method: one of fd|sys|no.
[... 118 lines cut ...]
  norecursedirs (args)     directory patterns to avoid for recursion
  testpaths (args)         directories to search for tests when no files or directories are given in the command line.
  usefixtures (args)       list of default fixtures to be used with this project
  python_files (args)      glob-style file patterns for Python test module discovery
  python_classes (args)    prefixes or glob names for Python test class discovery
  python_functions (args)  prefixes or glob names for Python test function and method discovery
  xfail_strict (bool)      default for the strict parameter of xfail markers when not given explicitly (default: False)
  doctest_optionflags (args) option flags for doctests
  addopts (args)           extra command line options
  minversion (string)      minimally required pytest version
  rsyncdirs (pathlist)     list of (relative) paths to be rsynced for remote distributed testing.
  rsyncignore (pathlist)   list of (relative) glob-style paths to be ignored for rsyncing.
  looponfailroots (pathlist) directories to check for changes

environment variables:
  PYTEST_ADDOPTS           extra command line options
  PYTEST_PLUGINS           comma-separated plugins to load during startup
  PYTEST_DEBUG             set to enable debug tracing of pytest's internals

to see available markers type: pytest --markers
to see available fixtures type: pytest --fixtures
(shown according to specified file_or_dir or current dir if not specified)

```
Since this doesn't seem to allow for the -a option from the Makefile command, I'm wondering was the '-a' a typo, or do I need to install some other version of py.test/pytest?

larsoner, 2017-08-28:
Yes you should update to the latest. It's pure Python so hopefully not too difficult to update

agramfort, 2017-08-28:
@massich this issue has now raised twice (before privately by @britta-wstnr)

any idea how to prevent this from happening again?shall we recommend / document something else?

britta-wstnr, 2017-08-28:
For me, installing the package ``pytest-attrib`` did the trick.

agramfort, 2017-08-28:
yes but it took you some time to find out :)

britta-wstnr, 2017-08-28:
Indeed :)

massich, 2017-08-28:
Yes to use nose attributes the pytest-attributes is needed. 

_Sent from my OnePlus ONEPLUS A5000 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

massich, 2017-08-28:
Using the pytest-attrib plugging helps to the transition between nose and pytest. But at some point when should get rid not only from nose dependency but also the way the tests were written (in this case labeled) 

_Sent from my OnePlus ONEPLUS A5000 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

agramfort, 2017-08-28:
> But at some point when should get rid not only from nose dependency but
also the way the tests were written (in this case labeled)

+10

smgutstein, 2017-08-29:
Thanks. Installing pytest-attrib was the missing plugin for me too.

agramfort, 2017-08-29:
this really needs to be documented

larsoner, 2017-08-29:
Can we just switch to standard pytest fixtures instead? We need to do that eventually anyway

agramfort, 2017-08-29:
+10

larsoner, 2017-09-02:
Fix is in #4523

PR #4541 (2017-09-12): MRG: Minor fixes to testing framework
This PR:

1. Gets rid of some `nose` cruft
2. Moves to standard `pytest` slow-test markers
3. Fixes our decorators for `sklearn` that had incorrect versions
4. Makes AppVeyor a little bit nicer

Closes #4518.
closed 2017-09-12 by larsoner