mne-tools/mne-python #4522: Empty comment in empty Evoked

cmmoenne, 2017-08-30:
Some of my epochs files have zero trials in some conditions. In that specific case, empty Evoked objects have the comment field empty (''). The EpochsFiF object for the condition (epochs[label]) has the _name field also empty ('').

Thus:

`epochs.event_id.keys()` has _condition_, epochs[_condition_]._name is empty when n_events = 0, and `epochs[_condition_].average().comment` is also empty when n_epochs = 0.

I'm not sure when the comment is lost, but (perhaps) keep the comment would be nice to track empty epochs/evokeds.

agramfort, 2017-08-30:
please provide a gist on sample data so we can replicate

thanks

cmmoenne, 2017-08-30:
sure! (and sorry for forgot to add a test file from the begining)

https://gist.github.com/cmmoenne/3b02ce03f9d55c649be2ce176d0d6ca1

agramfort, 2017-08-31:
it's because _name is a property dynamically generated from event_id :

https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L823

so if no event is present it's empty. How should we change the logic?

cmmoenne, 2017-08-31:
One posibility is check in `__getitem__` that `epochs.event_id` has at least the requested item before returning the new epoch (https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L1387).

As the item is in the `event_id` of the indexed epoch object, if `len(epochs.event_id) == 0` at the end of `__getitem__` we could add the requested item from the original. Thus, _name will use the inserted item which in turn will be passed to the evoked instance to fill the comment.

agramfort, 2017-09-04:
can you give a try to a PR to see if this works?

cmmoenne, 2017-09-04:
While coding the PR I got confused with the expected behavior of `__getitem__`. In one hand, if we replace:
```python
epochs.event_id = dict((k, v) for k, v in epochs.event_id.items()
                               if v in epochs.events[:, 2])
```
with:
```python
epochs.event_id = dict((k, epochs.event_id[k]) for k in _hid_match(epochs.event_id, item))
```
we always maintain the requested items, independent of the number of elements in epochs[item]. Note that the user will be able to select an empty epoch from an empty Epoch and so on (not sure if that is desirable).

However, if the idea is maintain only the labels of the items that are found in epochs, `__getitem__` is doing the right thing. In this case, we would only like to throw a warning `if len(epochs.event_id) == 0` (empty selection of item(s) %s, or something like that).

Perhaps, make the user aware is sufficient in this case. What do you think @agramfort ?

agramfort, 2017-09-12:
>
> However, if the idea is maintain only the labels of the items that are
> found in epochs, __getitem__ is doing the right thing.
>

yes it's the original idea.

> In this case, we would only like to throw a warning if
> len(epochs.event_id) == 0 (empty selection of item(s) %s, or something
> like that).
>
yes agreed.

> Perhaps, make the user aware is sufficient in this case. What do you think
> @agramfort <https://github.com/agramfort> ?
>

yes I think warning / exception is a good approach.

closed 2017-09-14 by cmmoenne