mne-tools/mne-python #4532: Does current MNE-Python pass all unit tests?

smgutstein, 2017-09-04:
I can't get my version of MNE python to pass unittests. I'm using (I believe) the current version:

```
me@Bedrock:~/mne-python$ git log -1 
commit 525dab575f6003df9441a86358de5415c8e2eb1e
Author: jona-sassenhagen <jona.sassenhagen@gmail.com>
Date:   Mon Sep 4 09:02:30 2017 +0200
```

However, I get the following result:

```
=================================================================
2 tests deselected 
1 failed, 614 passed, 45 skipped, 2 deselected, 149 warnings in 455.57 seconds =================================================================
```
The test that fails is:

    mne/simulation/tests/test_raw.py .F.

It fails at:
 ```
   mne/simulation/tests/test_raw.py:235: in test_simulate_raw_bem
        assert_true(med_diff < tol, msg='%s: %s' % (bem, med_diff))
```
I don't truly understand what is being tested here, but when I print out some extra info, I find that

    tol = 1
    med_diff = 1.1321....
    (also, min_diff = 0.075 2& max_diff =4.4047)
Does this happen to anyone else? Should I worry?

larsoner, 2017-09-04:
It passes over here, and the CIs have been happy. What is your `mne.sys_info`?

smgutstein, 2017-09-05:
>>> mne.sys_info()
Platform:      Linux-4.4.0-93-generic-x86_64-with-Ubuntu-16.04-xenial
Python:        2.7.12 (default, Nov 19 2016, 06:48:10)  [GCC 5.4.0 20160609]
Executable:    /usr/bin/python
CPU:           x86_64: 12 cores
Memory:        62.8 GB

mne:           0.15.dev0
numpy:         1.12.0 {lapack=openblas, blas=openblas}
scipy:         0.17.1
matplotlib:    2.0.0b3+1814.g3cb79ed

sklearn:       0.18
nibabel:       2.1.0
mayavi:        Not found
pycuda:        2016.1.2
skcuda:        0.5.2
pandas:        0.18.1

agramfort, 2017-09-05:
Indeed I can replicate... with this sys_info

```
In [2]: mne.sys_info()
Platform:      Darwin-16.6.0-x86_64-i386-64bit
Python:        3.5.3 | packaged by conda-forge | (default, Feb 10 2017,
07:09:50)  [GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)]
Executable:    /Users/alex/anaconda/envs/py35/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.15.dev0
numpy:         1.12.1 {lapack=/Users/alex/anaconda/envs/py35/lib,
blas=/Users/alex/anaconda/envs/py35/lib}
scipy:         0.19.1
matplotlib:    2.0.2

sklearn:       0.20.dev0
nibabel:       2.2.0dev
mayavi:        4.5.0
pycuda:        Not found
skcuda:        Not found
pandas:        0.20.1
```

but not this one for which it works:

```
In [2]: mne.sys_info()
Platform:      Darwin-16.6.0-x86_64-i386-64bit
Python:        3.6.2 |Continuum Analytics, Inc.| (default, Jul 20 2017,
13:14:59)  [GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)]
Executable:    /Users/alex/anaconda/envs/mne/bin/python
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.15.dev0
numpy:         1.13.1 {blas=mkl_intel_lp64, lapack=mkl_intel_lp64}
scipy:         0.19.1
matplotlib:    2.0.2

sklearn:       0.19.0
nibabel:       2.1.0
mayavi:        4.5.0
pycuda:        Not found
skcuda:        Not found
pandas:        0.20.3```

larsoner, 2017-09-05:
So mkl 3.6 passes and lapack 3.5 fails. I'll try to replicate.

agramfort, 2017-09-05:
ok it's a numpy version issue

larsoner, 2017-09-05:
Yeah could be 1.13 vs 1.12, I'll look this week

PR #4539 (2017-09-07): MRG: Fix tol
Closes #4532.

@smgutstein can you see if tests pass for you now?
closed 2017-09-07 by agramfort