mne-tools/mne-python #4533: BUG in _make_volume_source_space()

d-van-de-velden, 2017-09-05:
Dear mne-python, 

I am setting up volume source spaces for my study to perform MFT later on. Furthermore I am aiming for a statistical group study using Spatio temporal Clustering Test.
On my way of doing this I encountered two problems, which can be - in my opinion - be solved by gathering the correct information of neighbors for each vertice.
Inside the routine for setting up a volume source space (inside the make_volume_source space, source_space file line 1776 and following) it is declared to Figure out the 6-neighborhood and later 26, neighborhood for each vertice. Unfortunately after setting up an empty aray of the correct form and filling it with values of '-1', the function just creates 'True' and 'False' statements for the 'neigh'-array without assigning any vertice idx to the neighborhood array of each array. At last the NOT 'inuse' vertices are removed from the 'neigh'-array, which is than passed to the volume source space dictionary, still filled with values of '-1'.
I think there is something wrong with the functionality of this part of the function. It is not harmful to the correct result of the volume source space, but it would produce very useful information for later processing steps.
Is there a chance that this part is going to be 'corrected' with the next update of mne-python ?

Greetings,
        Daniel van de Velden

larsoner, 2017-09-06:
It would probably be simple enough to add a function to (re)compute the neighbors for a volume grid source space, as it would be backward- and forward-compatible (and probably also very fast since we should be able to vectorize it). It probably won't make it into the next (imminent) release, but hopefully the one after that.

agramfort, 2017-09-06:
please share any snippet of code. it's hard for me to understand your
problem without code I can read and run to see the problem.

do you have an MFT implementation compatible with mne-python?

d-van-de-velden, 2017-09-07:
I am sorry for that, A. Gramfort. Here is a short script that hopefully gives you a better understanding.

```
#!/usr/bin/env python2
# -*- coding: utf-8 -*-
"""
Author:   Daniel van de Velden                  mail: daniel.vandevelden@yahoo.de
Created on Tue Sep 5th, 01:15:11 2017
"""

from mne import  setup_volume_source_space
from mne.datasets import sample

data_path = sample.data_path()
subjects_dir = data_path + '/subjects'
subj = 'sample'
aseg_fname = subjects_dir + '/sample/mri/aseg.mgz'

# setup a volume source space of the sample brain
sphere = (0, 0, 0, 120)
vol_src = setup_volume_source_space(subj, mri=aseg_fname, sphere=sphere,
                                     subjects_dir=subjects_dir)

print 'Vol. SrcSp has ', vol_src[0]['neighbor_vert'].shape[0], 'vertices.'
print 'Each with ', vol_src[0]['neighbor_vert'].shape[1] ,' neighbors.'
print 'Expected content of this array (vertice indexes of neighborhood vertices):'
print """array([[3122, 5233, 1234, ..., 3123, 4643, 2344],
       [ and so on ....]])"""
print "Real content, regardless of the input of 'setup_volume_source_space'"
print vol_src[0]['neighbor_vert']
```
My investigation so far suggests the mistake to be inside the '/mne/source_space.py'-file. Function '_make_volume_source_space', line 1712:

```
def _make_volume_source_space(surf, grid, exclude, mindist, mri=None,
                              volume_label=None, do_neighbors=True, n_jobs=1):
    """Make a source space which covers the volume bounded by surf."""

                             [...]

#==============================================================================
#     START Added by Daniel van de Velden
#==============================================================================
    print '\n Doing Neighbors... \n'
#==============================================================================
#     END Added by Daniel van de Velden
#==============================================================================

    k = np.arange(npts)
    neigh = np.empty((26, npts), int)
    neigh.fill(-1)

    # Figure out each neighborhood:
    # 6-neighborhood first
    idxs = [z > minn[2], x < maxn[0], y < maxn[1],
            x > minn[0], y > minn[1], z < maxn[2]]
    print idxs
    offsets = [-nplane, 1, nrow, -1, -nrow, nplane]
    for n, idx, offset in zip(neigh[:6], idxs, offsets):
        n[idx] = k[idx] + offset
    print n
    # Then the rest to complete the 26-neighborhood

    # First the plane below
    idx1 = z > minn[2]
    print '', idx1,' idx1'
    idx2 = np.logical_and(idx1, x < maxn[0])
    print '', idx2,' idx2'
    neigh[6, idx2] = k[idx2] + 1 - nplane
    print neigh[6]
    print neigh[6].shape
    idx3 = np.logical_and(idx2, y < maxn[1])
    print 'neigh 7 before:', neigh[7]
    neigh[7, idx3] = k[idx3] + 1 + nrow - nplane
    print 'neigh 7 after:', neigh[7]
    idx2 = np.logical_and(idx1, y < maxn[1])
    neigh[8, idx2] = k[idx2] + nrow - nplane

    idx2 = np.logical_and(idx1, x > minn[0])
    idx3 = np.logical_and(idx2, y < maxn[1])
    neigh[9, idx3] = k[idx3] - 1 + nrow - nplane
    neigh[10, idx2] = k[idx2] - 1 - nplane
    idx3 = np.logical_and(idx2, y > minn[1])
    neigh[11, idx3] = k[idx3] - 1 - nrow - nplane

    idx2 = np.logical_and(idx1, y > minn[1])
    neigh[12, idx2] = k[idx2] - nrow - nplane
    idx3 = np.logical_and(idx2, x < maxn[0])
    neigh[13, idx3] = k[idx3] + 1 - nrow - nplane

    # Then the same plane
    idx1 = np.logical_and(x < maxn[0], y < maxn[1])
    neigh[14, idx1] = k[idx1] + 1 + nrow

    idx1 = x > minn[0]
    idx2 = np.logical_and(idx1, y < maxn[1])
    neigh[15, idx2] = k[idx2] - 1 + nrow
    idx2 = np.logical_and(idx1, y > minn[1])
    neigh[16, idx2] = k[idx2] - 1 - nrow

    idx1 = np.logical_and(y > minn[1], x < maxn[0])
    neigh[17, idx1] = k[idx1] + 1 - nrow - nplane

    # Finally one plane above
    idx1 = z < maxn[2]

    idx2 = np.logical_and(idx1, x < maxn[0])
    neigh[18, idx2] = k[idx2] + 1 + nplane
    idx3 = np.logical_and(idx2, y < maxn[1])
    neigh[19, idx3] = k[idx3] + 1 + nrow + nplane

    idx2 = np.logical_and(idx1, y < maxn[1])
    neigh[20, idx2] = k[idx2] + nrow + nplane

    idx2 = np.logical_and(idx1, x > minn[0])
    idx3 = np.logical_and(idx2, y < maxn[1])
    neigh[21, idx3] = k[idx3] - 1 + nrow + nplane
    neigh[22, idx2] = k[idx2] - 1 + nplane
    idx3 = np.logical_and(idx2, y > minn[1])
    neigh[23, idx3] = k[idx3] - 1 - nrow + nplane

    idx2 = np.logical_and(idx1, y > minn[1])
    neigh[24, idx2] = k[idx2] - nrow + nplane
    idx3 = np.logical_and(idx2, x < maxn[0])
    neigh[25, idx3] = k[idx3] + 1 - nrow + nplane

                             [...]

    # Omit unused vertices from the neighborhoods
    logger.info('Adjusting the neighborhood info...')
    # remove non source-space points
    log_inuse = sp['inuse'] > 0
    
    print 'neigh before:', neigh
    neigh[:, np.logical_not(log_inuse)] = -1
    print neigh.shape
    print 'neigh after -1:', neigh
    print neigh.shape
    # remove these points from neigh
    vertno = np.where(log_inuse)[0]
    sp['vertno'] = vertno
    old_shape = neigh.shape
    neigh = neigh.ravel()
    checks = np.where(neigh >= 0)[0]
    removes = np.logical_not(np.in1d(checks, vertno))
    neigh[checks[removes]] = -1
    neigh.shape = old_shape
    neigh = neigh.T
    # Thought we would need this, but C code keeps -1 vertices, so we will:
    # neigh = [n[n >= 0] for n in enumerate(neigh[vertno])]
    sp['neighbor_vert'] = neigh
```

With that, 'neigh' is being passed to the VolumeSourceSpace dictionary filled with values of '-1'.

jdammers, 2017-09-07:
@agramfort, cc @larsoner, @d-van-de-velden: We will be happy to share the (MNE-) python version of MFT (if you are interested) once we are happy with our testy we do. We currently run a full MFT based source analysis on a study we did including spatio-temporal permutations test to find active brain regions in full volumetric source spaces across subjects. As @d-van-de-velden mentioned, we face some problems with regard to i) morphing the volumetric source space to fsaverage and ii) running the statistical tests on the volumetric data (definition of the neighbourhood as described above).

agramfort, 2017-09-08:
regarding i) what did you do? use FS, FSL, SPM?

for ii) maybe have a look at
https://gist.github.com/agramfort/a3e3139a1f64315885c923a57e541e7a ?

see the use of grid_to_graph to extract the neighborhood structure.

larsoner, 2017-09-08:
We should add or wrap `grid_to_graph` to make a neighbor function like we have for surface source spaces

d-van-de-velden, 2017-09-08:
Regarding i) we tried to keep the whole processing (registration and morphing) in the mne-python and/or FreeSurfer enviroment/funtionality. Using:
```
morph_mat = mne.compute_morph_matrix(subject_from, subject_to,
                                     stc_from.vertices, vertices_to,
                                     subjects_dir=subjects_dir)
stc_to_2 = mne.morph_data_precomputed(subject_from, subject_to,
                                      stc_from, vertices_to, morph_mat)
```
and trying to modify it to accept volume source spaces and morph it correctly. Unfortunately there are still some mistakes.
Respectively ii) we haven't tackled this problem, since we want to finish the morphing process first. But thanks for the hint. It would be nice to have `grid_to_graph` implemented to gather the neighborhood relations in the volume source space.

agramfort, 2017-09-09:
have a look at :

https://github.com/mne-tools/mne-python/pull/4544

agramfort, 2017-11-15:
@d-van-de-velden can we close this? did #4544 help?

d-van-de-velden, 2017-11-15:
Thank you, Mr. Gramfort. The example helped a lot. Respectivly the neighborhood connections the solution is fine.
However we stepped back from mne-python's registration + morphing approach and thought of an alternative way. 
If you are intrested in the resulting source code you can leave this issue open, otherwise you can close it.
Greetings,
Daniel

agramfort, 2017-11-15:
sure please share any progress and if you have an implementation of MFT we
can try it would be great.

larsoner, 2018-04-02:
Closing this for now because I think the issue mentioned here (neighbors) has been solved by #4544, but feel free to comment further on volume reg / morphing stuff here or in new issues.

closed 2018-04-02 by larsoner