mne-tools/mne-python #4534: Capitalization in axis labels

bgagl, 2017-09-05:
Hey guys 
amazing stuff your are doing ...
just wondered if it would be possible to switch the default for axis labeles to first letter capitalized (e.g. from time [ms] to Time [ms])? 

Would be great as it is a requirement for several journals ... (Nature: Lettering in figures (labelling of axes and so on) should be in lower-case type, with the first letter capitalized and no full stop.)

Cheers
Ben

dengemann, 2017-09-05:
Hey Benjamin (it's you I hope), thanks!
I think the better thing to watch our for is how to edit matplotlib figures.
The idea would be that you get the axis text objects for the titles and
just replace them.
See here:
https://stackoverflow.com/questions/11244514/modify-tick-label-text
Keep in mind that our plotters are in the end providing convenient short
cuts for matplotlib but do not and cannot replace matplotlib.

Denis

bgagl, 2017-09-06:
Hi Denis, 
yes its me ;) 
ok I see. Unfortunately, I like e.g. the plot_joint function very much and as far as I can see you would recommend to take the code and customize the figure directly with the code itself.
Cheers

dengemann, 2017-09-06:
But you can do that! All plotting functions return matplotlib figure
handles. You can edit any detail of a figure post-hoc using matplotlib.

jona-sassenhagen, 2017-09-06:
Maybe we should do a tutorial for styling default figure outputs.
(I'm comfortable with matplotlib, but arguably it's pretty arcane to use. MNE is a lot more beginner friendly than matplotlib!)

On the other hand, if journals require capitalisation, why should we return all-lower case in the first place? We're implicitly making a choice either way, so why the one to return lower case?

dengemann, 2017-09-06:
We can do both :)

jona-sassenhagen, 2017-09-06:
Who is "we" :D

But yes, you're right.

agramfort, 2017-09-06:
> On the other hand, if journals require capitalisation, why should we
return all-lower case in the first place? We're implicitly making a choice
either way, so why the one to return lower case?

same feeling. Can you build case for capitalisation ? then we could change
all our defaults.

cbrnr, 2017-10-19:
And while we're at it, "Time [ms]" is wrong, units should be in parens and not brackets, so it should be "Time (ms)" (not that I think MNE is using brackets, but it might be worth checking).

PR #5079 (2018-04-03): MRG: Unify axis labeling
Closes #4534.

This PR:

- Unifies axis labeling to "Time (s)" form from others like "time [sec]". Turns out there were a lot of places we already used the former standard so the changeset wasn't so bad.
- Unifies handling of unit naming in ICA property plots to fit with e.g. `raw.plot_psd()`.
- Unifies `sphx_glr_*`-based linking rather than custom `tut_` ones to make things more standard.
closed 2018-04-03 by agramfort