mne-tools/mne-python #4538: colormap of individual maps in evoked.plot_joint()

timmidee, 2017-09-06:
This one will probably end up with @jona-sassenhagen .

Is appears that when all data are positive at a time-point passed for the argument "times", the colormap of the individual topography at that time-point is set to "reds". Which I believe is standard behavior for the function that plots the topomap. It doesn't adhere to the colorbar though. It is particularly striking pre t=0 in the attached image.

Explicitly passing the colormap as an argument in topomap_args helps, but perhaps this can be handled more elegantly? It took me a while to notice. Sometimes it can be fairly obscured (like all electrodes around a strong peak being a "little too red").

[image]

larsoner, 2017-09-06:
Yeah the `plot_joint` should look at all data points, the topomap ones individually, to determine the colormap to use for all of the topomaps

jona-sassenhagen, 2017-09-06:
@jaeilepp do you have time? Or should I do it?

Or maybe you want to do it yourself, @timmidee

agramfort, 2017-09-06:
@jaeilepp <https://github.com/jaeilepp> is not working with me anymore but
maybe he needs a break with his new job :)

jona-sassenhagen, 2017-09-07:
@timmidee I can't replicate the problem. Can you share the evoked dataset and the precise call to plot_joint?

timmidee, 2017-09-08:
`times = (-.190,-.030,-.010,.100,.275,.460)

topomap_args = dict(average = .05,cmap='RdBu_r')

sk_sacSinCoef.plot_joint(times = times, topomap_args=topomap_args,
                           title='rERP estimated effect of continuous covariate fixIncSac SIN',
                           ts_args= None);`

Note the difference at -10 and 275ms when omitting the cmap from topomap_args.
This one does not mess up the colorbar, but that happens when the first map is set to cmap='reds'.
I assume that will fix itself once all maps get the same colormap ;-)

[weird_map-ave.fif.zip](https://github.com/mne-tools/mne-python/files/1288413/weird_map-ave.fif.zip)

[image]
[image]

jona-sassenhagen, 2017-10-20:
Hm. I still can't replicate this. I get the 1st plot regardless of if I include a cmap or not. Has this somehow been fixed already?

jona-sassenhagen, 2017-10-20:
The crucial lines are probably:

```
    norm = min(data) >= 0
    vmin, vmax = _setup_vmin_vmax(data, vmin, vmax, norm)
    if cmap is None:
        cmap = 'Reds' if norm else 'RdBu_r'
```
here
https://github.com/mne-tools/mne-python/blob/9ac6547284bbd73b4c18d7d7031042585388005c/mne/viz/topomap.py#L606

timmidee, 2017-10-22:
I can't try right now, but looking at the code you cite (with the risk of saying something stupid here):

If this is done on a map-to-map basis then isn't that the problem?
I.e. the overarching function (plot_joint()) should look at the data for _all_ maps and then consistently pass the colormap to plot_topomap()?

Or perhaps I'm missing the structure of what function calls and/or passes what here. In that case let me know. I'll give this another try with the latest version of MNE when I have easy access to my work machine again.

larsoner, 2018-04-04:
Closing as it sounds like this has been fixed, but @timmidee if you still see this on `master` please reopen

closed 2018-04-04 by larsoner