mne-tools/mne-python #4540: Decoding error loading GDF

Morgan243, 2017-09-08:
Trying to load the GDF files from BCI IV 2a (similar to #4253), but I get decoding errors in both Python 2 and 3. I'm able to load this data in BioSig and SigViewer, so the files appear to be valid. Maybe I'm doing something wrong?

Test code:

```python
import mne
p = './A01E.gdf'
raw_edf = mne.io.read_raw_edf(p, stim_channel=None)`
```

In Python 2:

```
Extracting edf Parameters from ./A01E.gdf...

-------------------------------------------------------------------
UnicodeDecodeError                Traceback (most recent call last)
<ipython-input-20-450027490c39> in <module>()
      3 p = './A01E.gdf'
      4 
----> 5 raw_edf = mne.io.read_raw_edf(p, stim_channel=None)

/home/morgan/py2_venv/lib/python2.7/site-packages/mne/io/edf/edf.pyc in read_raw_edf(input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
    606     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
    607                   stim_channel=stim_channel, annot=annot, annotmap=annotmap,
--> 608                   exclude=exclude, preload=preload, verbose=verbose)

/home/morgan/py2_venv/lib/python2.7/site-packages/mne/io/edf/edf.pyc in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)

/home/morgan/py2_venv/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
  Extracting edf Parameters from ./A01E.gdf...

-------------------------------------------------------------------
UnicodeDecodeError                Traceback (most recent call last)
<ipython-input-1-450027490c39> in <module>()
      3 p = './A01E.gdf'
      4 
----> 5 raw_edf = mne.io.read_raw_edf(p, stim_channel=None)

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in read_raw_edf(input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
    606     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
    607                   stim_channel=stim_channel, annot=annot, annotmap=annotmap,
--> 608                   exclude=exclude, preload=preload, verbose=verbose)

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)

~/venv/lib/python3.6/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
     77         input_fname = os.path.abspath(input_fname)
     78         info, edf_info = _get_edf_info(input_fname, stim_channel, annot,
---> 79                                        annotmap, eog, misc, exclude, preload)
     80         logger.info('Creating Raw.info structure...')
     81         _check_update_montage(info, montage)

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in _get_edf_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    315 
    316         day, month, year = [int(x) for x in re.findall('(\d+)',
--> 317                                                        fid.read(8).decode())]
    318         hour, minute, sec = [int(x) for x in re.findall('(\d+)',
    319                                                         fid.read(8).decode())]

UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 3: invalid start byte  708 
    709 

/home/morgan/py2_venv/lib/python2.7/site-packages/mne/io/edf/edf.pyc in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
     77         input_fname = os.path.abspath(input_fname)
     78         info, edf_info = _get_edf_info(input_fname, stim_channel, annot,
---> 79                                        annotmap, eog, misc, exclude, preload)
     80         logger.info('Creating Raw.info structure...')
     81         _check_update_montage(info, montage)

/home/morgan/py2_venv/lib/python2.7/site-packages/mne/io/edf/edf.pyc in _get_edf_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    315 
    316         day, month, year = [int(x) for x in re.findall('(\d+)',
--> 317                                                        fid.read(8).decode())]
    318         hour, minute, sec = [int(x) for x in re.findall('(\d+)',
    319                                                         fid.read(8).decode())]

UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 3: ordinal not in range(128)
```

And in Python 3:

```
Extracting edf Parameters from ./A01E.gdf...

-------------------------------------------------------------------
UnicodeDecodeError                Traceback (most recent call last)
<ipython-input-1-450027490c39> in <module>()
      3 p = './A01E.gdf'
      4 
----> 5 raw_edf = mne.io.read_raw_edf(p, stim_channel=None)

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in read_raw_edf(input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
    606     return RawEDF(input_fname=input_fname, montage=montage, eog=eog, misc=misc,
    607                   stim_channel=stim_channel, annot=annot, annotmap=annotmap,
--> 608                   exclude=exclude, preload=preload, verbose=verbose)

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)

~/venv/lib/python3.6/site-packages/mne/utils.py in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in __init__(self, input_fname, montage, eog, misc, stim_channel, annot, annotmap, exclude, preload, verbose)
     77         input_fname = os.path.abspath(input_fname)
     78         info, edf_info = _get_edf_info(input_fname, stim_channel, annot,
---> 79                                        annotmap, eog, misc, exclude, preload)
     80         logger.info('Creating Raw.info structure...')
     81         _check_update_montage(info, montage)

~/venv/lib/python3.6/site-packages/mne/io/edf/edf.py in _get_edf_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    315 
    316         day, month, year = [int(x) for x in re.findall('(\d+)',
--> 317                                                        fid.read(8).decode())]
    318         hour, minute, sec = [int(x) for x in re.findall('(\d+)',
    319                                                         fid.read(8).decode())]

UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 3: invalid start byte
```

agramfort, 2017-09-08:
can you share the file here in a .zip file? or share a dropbox link?

Morgan243, 2017-09-08:
I'm not sure I can share them legally, but they are freely available. You have to provide your email and get a generated password. Go here to do that:
http://bbci.de/competition/iv/#download

Then downloads/descriptions are here. The files I'm using are for task 2a
http://bbci.de/competition/iv/download/

Thank you!

Morgan243, 2017-09-09:
Updating to the current master branch seems to fix it

closed 2017-09-09 by Morgan243

agramfort, 2017-09-09:
great. Thanks for checking.