mne-tools/mne-python #4551: `high is out of bounds for int64` in cluster permutation test

nbara, 2017-09-13:
I've been getting an error in cluster permutation tests when applied on single-trial data, when the number of trials exceeds ~60

FWICT this is due to `max_perms` being too large in cluster_level.py : 
https://github.com/mne-tools/mne-python/blob/cd24596e14383dc978d754a51f8fe9fabbd76b21/mne/stats/cluster_level.py#L761-L768

I can reproduce the error when running [this MNE tutorial](https://github.com/mne-tools/mne-python/blob/master/tutorials/plot_stats_cluster_1samp_test_time_frequency.py), when rejecting no epochs (`Reject=None`). In that case `epochs_power` has shape (72, 16, 49), which isn't extreme, but I'm still getting the following error:

<details><summary>Log output</summary>

```python
Traceback (most recent call last):
  File "/Users/nicolas/Toolboxes/mne-python-nbara/mne/stats/cluster_level.py", line 1163, in permutation_cluster_1samp_test
    buffer_size=buffer_size)
  File "<string>", line 2, in _permutation_cluster_test
  File "/Users/nicolas/Toolboxes/mne-python-nbara/mne/utils.py", line 726, in verbose
    return function(*args, **kwargs)
  File "/Users/nicolas/Toolboxes/mne-python-nbara/mne/stats/cluster_level.py", line 768, in _permutation_cluster_test
    orders = rng.choice(max_perms, n_permutations - 1)
  File "mtrand.pyx", line 1147, in mtrand.RandomState.choice (numpy/random/mtrand/mtrand.c:18069)
  File "mtrand.pyx", line 975, in mtrand.RandomState.randint (numpy/random/mtrand/mtrand.c:16083)
ValueError: high is out of bounds for int64
```

</details>

agramfort, 2017-09-13:
don't do 2**60 permutations. limit n_permutations to lets say 10000 for a
p-value threshold at 0.01

nbara, 2017-09-13:
Mmh but doesn't `n_permutations` default to 1024 in `permutation_cluster_1samp_test` &co?

In the code snippet above, `max_perms` is only a function of the number of samples (in my case, trials)

mmagnuski, 2017-09-13:
@agramfort - the problem seems to be different, the realized premutations are chosen from all possible permutations (as an int that is later transformed to its binary representation) and it seems that if there are too many possible permutations `rng.choice` gives errors.

larsoner, 2017-09-13:
I see my "clever trick" is really a "bug" sometimes :)

We can go back to using resampling with replacement in the non-exact case. (My code changed it to be sampling without replacement.) Sound good? If so I can make a PR

agramfort, 2017-09-13:
you should not allow replacement in theory AFAIK

larsoner, 2017-09-13:
Okay I think there is another way to do it that should be safe enough

PR #4552 (2017-09-13): MRG: Fix permutations without replacement
In practice the prevention of collisions is probably a bit overkill, but this should ensure we sample without replacement. But even in the intermediate range `20 < n_samples < 64` the `rng.choice` call was a bit rough on memory usage, so this fixes that, too.

Closes #4551.

@nbara can you see if it fixes your problem?
closed 2017-09-13 by larsoner