mne-tools/mne-python #4555: Proposed taxonomy: decoding/encoding; generative/discriminative; forward/backward

kingjr, 2017-09-13:
As discussed off-line, here is a to-be-debated proposal for operational definitions:

Supervised ML analyses to neuroimaging tend to cluster into two groups: forward generative encoding models versus backward discriminative models. The corresponding terms have thus been used interchangeably (e.g. Haufe et al 2014). However, they actually refer to independent properties of any neuroimaging analysis.

* **Decoding** and **encoding** are representational concepts. In neuroscience, they refer to the directionality of the mapping between neuronal data on one hand, and other variables on the other hand: experimentally manipulated events (e.g., stimulus characteristics such as the orientation of a visual grating), non-neuronal observations (e.g., overt responses), or parameters of cognitive models (e.g. confidence of a motor decision of an ideal observer). Decoding consists in *predicting* such variables from neuronal activity: P(Variable|Neuronal Activity). Encoding refers to predicting neuronal activity from this variable: P(Neuronal Activity|Variable).

[image]

* **Backward** and **forward** models indicate whether the encoding or decoding model is assessed in the same way it has been built or optimized. For example, a logistic regression fitted to maximize the discriminability of “faces” versus “houses” and scored precisely on that ability is a forward model: i.e. it is fitted and scored in a decoding fashion. However, a model fitted to maximize the variance induced by faces and houses, and subsequently assessed on its ability to predict the visual stimulus given brain activity is a backward encoding model.

* **Generative** and **discriminative** models are statistical concepts. Generative models estimate the joint distribution between two observables X and Y: P(X, Y). Discriminative models estimate the conditional probability between two observables: P(Y|X). A discriminative model can always be derived from a generative model using Bayesian rule, but not vice versa. Discriminative models therefore make fewer assumptions, and can thus be more robust to a particular task but less informative and less general than their generative counterparts.

The present proposal implies that these 3 dimensions are not mutually exclusive. 
For example, fitting an SVC to predict a stimulus category from a continuous BOLD response consists in fitting a forward discriminative decoder. Fitting an SVM to predict whether a neuron should spike given the contrasts of a stimulus' pixels consists in fitting a forward discriminative encoder. For some models, and in particular for univariate analyses, these distinctions can be irrelevant. For example, a simple ordinary least-square regression (equivalent to LDA and ANOVA in binary cases) will find the same model whether fit in its encoding and decoding directions. Additionally, it is important to highlight that the coding framework is not a mind / body problem. Autoregressive models, that are optimized to predict future brain activity can be thought of as encoding models from past neuronal data (as opposed to current sensory stimulation, or future motor actions).

The limits with this proposal
* it does not integrate other linear approaches: e.g. hyperalignment
* forward and backward are used slightly differently in control theory
* forward and inverse modeling for MEEG source modeling may not well fit with the present forward/backward distinction.

Please comment if you disagree or if you think we should add additional info. I'll update the proposal accordingly until we reach an agreement.

agramfort, 2017-09-14:
I think it is really clear

> it does not integrate other linear approaches: e.g. hyperalignment

hyperalignment is a linear or non-inear transform to align brain data
in a common feature space. Once you have done this using some data
you are back to encoding / decoding but being able to predict for new
subjects.

> forward and backward are used slightly differently in control theory

if you say so.

> forward and inverse modeling for MEEG source modeling may not well fit
with the present forward/backward distinction.

M/EEG inverse modeling when linear is applying a spatial filter
informed by physics. The filter aims to highligh what deviates from zero /
from baseline

jasmainak, 2017-09-14:
sounds great but is the cat French? :)

jona-sassenhagen, 2018-05-12:
Can this be closed? :)

agramfort, 2018-05-12:
I guess so

larsoner, 2021-02-19:
I think we use this taxonomy nowadays

closed 2021-02-19 by larsoner