mne-tools/mne-python #4556: ML Documentation: what's missing

kingjr, 2017-09-13:
Here is a proposal in the todo for the MVPA/Decoding-Encoding/ML module.

- [ ] a brief tutorial on ML to introduce the key concepts (fitting, predicting, scoring, CV)
- [ ] a tutorial on model interpretation and its limits: when can you do patterns, what does it mean, what are the risks
- [ ] a tutorial to highlight the importance of supervised spatial filter for oscillatory activity
- [ ] a tutorial to highlight importance of supervised spatial filters for various types of artefacts
- [ ] a tutorial to do source decoding and its risks

Please comment and I'll update the proposal.

agramfort, 2017-09-14:
do you have already materials for some of this?

jasmainak, 2017-09-14:
It's more an API thing but I get a lot of questions which basically boil down to using 3D data in pipelines leading to weird shape mismatches. I guess some estimators can be pipelined in sequence but others cannot be. This needs to be probably clarified.

kingjr, 2017-09-14:
> do you have already materials for some of this?

Yes, in the form of ipynb for all of the above, but a bit too long as compared to the MNE tutorials. Shall I start opening PRs?

> It's more an API thing but I get a lot of questions which basically boil down to using 3D data in pipelines leading to weird shape mismatches. I guess some estimators can be pipelined in sequence but others cannot be. This needs to be probably clarified.

Ok that's easy, we can dedicate a tutorial to pipeline. I think it is an important an concept, especially with our tensor-like data structure.

agramfort, 2017-09-15:
let's not make PRs but maybe make a repo of 3-4 notebooks with an index.ipynb like here:

https://github.com/jakevdp/PythonDataScienceHandbook/tree/master/notebooks

then we take the figures from the notebook for the paper.