mne-tools/mne-python #4558: Custom colormap normalization in power plot

cbrnr, 2017-09-14:
I'd like to create a classic ERD/ERS time/frequency plot by computing a TFR (e.g. via `tfr_multitaper`). Power should be normalized to a baseline with `mode='percent'`. I want to set the limits to `vmin=-1` and `vmax=1.5` using `cmap='RdBu'` (because these are the values typically used in ERD/ERS maps). This works fine, but now I'd like to map the value 0 to white; negative values should linearly scale from -1 (red) to 0 (white), and positive values should linearly scale from 0 (white) to 1.5 (blue).

I found [this example](https://matplotlib.org/users/colormapnorms.html#custom-normalization-two-linear-ranges) on how to normalize colormaps using the `norm` argument of `pcolormesh` (the relevant line in MNE that produces the plot seems to be [here](https://github.com/mne-tools/mne-python/blob/master/mne/viz/topo.py#L309)). Is it possible to do this after the plot has been created (if so, how)? Or is this only possible by specifying the `norm` argument for `pcolormesh`?

agramfort, 2017-09-14:
no idea. Never did this.

cbrnr, 2017-09-14:
[This SO post](https://stackoverflow.com/a/20528097/1112283) describes how this could be done with a custom colormap - which would mean that no internal changes are necessary. However, I wonder if MNE should provide this out of the box to support creating standard ERD/ERS maps.

agramfort, 2017-09-14:
provide a gist by passing a custom cmap

cbrnr, 2017-09-14:
It's pretty straightforward, but I feel that generating a standard ERDS map (as per the settings used in the Pfurtscheller papers) could be useful for some people. That's why I initially created [a small package](https://github.com/cbrnr/erds), but later realized that this should really be done by MNE (because it already has all the stuff).

Here's basically what needs to be done:
```
vmin, vmax = -1, 1.5
vzero = abs(vmin) / (vmax - vmin)
cmap = plt.cm.RdBu
index_old = np.linspace(0, 1, cmap.N)
index_new = np.hstack([np.linspace(0, vzero, cmap.N / 2, endpoint=False),
                       np.linspace(vzero, 1, cmap.N / 2)])
cdict = {"red": [], "green": [], "blue": [], "alpha": []}
for old, new in zip(index_old, index_new):
    r, g, b, a = cmap(old)
    cdict["red"].append((new, r, r))
    cdict["green"].append((new, g, g))
    cdict["blue"].append((new, b, b))
    cdict["alpha"].append((new, a, a))

newcmap = LinearSegmentedColormap("erds", cdict)

power.plot([0], baseline=[-2.5, -0.5], mode="percent", vmin=vmin, vmax=vmax,
           cmap=(newcmap, False))
```

Of course this could be factored out into a function. Would it make sense to add this to the examples so that people actually find it?

larsoner, 2017-09-14:
Creating a custom colormap seems a bit of a hack when IIUC `norm` should do exactly what you want.

Maybe we should have a `norm` argument that:

1. If `centered`, will use `vmin` and `vmax` and make `0` at the center of the colormap
2. If anything else, pass to `matplotlib` (default `None`)

And then say in the docstring that `centered` can be useful for (wacky) asymmetrical plots where you still want zero at the center of the colormap.

cbrnr, 2017-09-14:
Yes, this would be really nice, but we'd still have to hack together a class that does the rescaling. I wish the Matplotlib devs implemented `colors.OffsetNorm` like they say on their website, this would completely solve this issue. Otherwise, if you think this is worth adding a new argument I'd be happy to implement it. If not, I'd argue for adding an example script on how to plot such ERDS maps.

larsoner, 2017-09-14:
They give code for how to do it with `MidpointNormalize`. Let's put it in `mne.fixes` under the assumption they'll add it eventually, then we're good to go with my API suggestion right?

agramfort, 2017-09-14:
my first reaction is make it an example with entry in the FAQ

larsoner, 2017-09-14:
To me it depends on how standard / often used it is, which I don't know

But it looks like you can modify it after the fact with `mesh.set_norm`, so we can go the example route pretty simply

cbrnr, 2017-09-14:
I know a lot of people who produce such ERDS maps, so I think MNE should make this task as straightforward as possible. I kind of tend toward adding an example too, since knowing a lot of people is no indicator of how standard or how often it is used :smile:. I'll submit a PR. Am I correct in assuming that the example should go in [this section](https://martinos.org/mne/stable/auto_examples/index.html#time-frequency-examples)?

larsoner, 2017-09-14:
Yeah that's the one

agramfort, 2017-09-14:
yes

PR #4566 (2017-09-19): [MRG] Add ERDS maps to time-frequency examples
Fix #4558. This example produces the following two plots:

[image]
[image]

This is already pretty OK, but there are two minor things that I'd like to change if it's not too much effort:

1. The x-axis should be in seconds.
2. The y-axis is weirdly spaced, ideally I'd like the frequency tick labels to be uniformly spaced.

Any easy ways to fix these things?
closed 2017-09-19 by larsoner