mne-tools/mne-python #4564: Import error from EEGLAB

pbrusco, 2017-09-14:
Hi! When importing epochs from EEGLAB, the function __init__ from io/eeglab/eeglab.py has a condition:
`if not isinstance(ep.eventtype, string_types):` and then treats the event as an array of events. 

The problem is that sometimes, the eventtype is a number (not an array), so, we add this as a patch:
```                
eventype = str(ep.eventtype)
if not isinstance(eventype, string_types):
```
And then we replace `ep.eventtype` for `eventype` and it seems to work. Any idea if this is the right way to solve it or we are missing something? 

Thanks!

jasmainak, 2017-09-14:
are you sure you are using the latest version of mne?

I think this was addressed here: https://github.com/mne-tools/mne-python/pull/3912. Or did I misunderstand your problem?

pbrusco, 2017-09-14:
Hi, sorry I didn't find the other issue, but yes, that was the problem.

Now, I tried with 0.15.dev0 and when the event is an int it's working. 

But, when the array contains ints, then it explodes again :). 

the solution I found, change 
```python
event_type = '/'.join(ep.eventtype.tolist())
``` 
for 
```python 
event_type = '/'.join(map(str, ep.eventtype.tolist()))
```

Thanks!

agramfort, 2017-09-15:
please share the problematic set file.

pbrusco, 2017-09-15:
Here it goes. 
[set_file.zip](https://github.com/mne-tools/mne-python/files/1306717/set_file.zip)

agramfort, 2017-09-17:
thanks for sharing.

please share a gist that produces the crash (saves us time)

also if you can send us a pull request so we clearly see the diff for your
changes it's even better.

thanks a lot

jona-sassenhagen, 2017-09-18:
In general ... @jasmainak do you have time to do this?

jasmainak, 2017-09-19:
see: https://github.com/mne-tools/mne-python/pull/4581/

closed 2017-09-20 by larsoner