mne-tools/mne-python #4572: Is Mayavi required for mne-python?

smgutstein, 2017-09-16:
I just updated to the latest version of mne python, but the `make test` failed :

```
$ make test
python setup.py build_ext -i
running build_ext
rm -f .coverage
py.test -m 'not ultraslowtest' mne
============================================================================== test session starts ===============================================================================
platform linux2 -- Python 2.7.12, pytest-3.2.1, py-1.4.34, pluggy-0.4.0
rootdir: /home/smgutstein/Downloads/temp9/mne-python, inifile: setup.cfg
plugins: xdist-1.20.0, forked-0.2, cov-2.5.1, attrib-0.1.3
collected 706 items / 5 errors                                                                                                                                                    

===================================================================================== ERRORS =====================================================================================
_____________________________________________________________________ ERROR collecting mne/gui/_coreg_gui.py _____________________________________________________________________
mne/gui/_coreg_gui.py:47: in <module>
    from mayavi.core.ui.mayavi_scene import MayaviScene
E   ImportError: No module named mayavi.core.ui.mayavi_scene
___________________________________________________________________ ERROR collecting mne/gui/_fiducials_gui.py ___________________________________________________________________
mne/gui/_fiducials_gui.py:10: in <module>
    from mayavi.core.ui.mayavi_scene import MayaviScene
E   ImportError: No module named mayavi.core.ui.mayavi_scene
___________________________________________________________________ ERROR collecting mne/gui/_kit2fiff_gui.py ____________________________________________________________________
mne/gui/_kit2fiff_gui.py:20: in <module>
    from mayavi.core.ui.mayavi_scene import MayaviScene
E   ImportError: No module named mayavi.core.ui.mayavi_scen
```e
____________________________________________________________________ ERROR collecting mne/gui/_marker_gui.py _____________________________________________________________________
mne/gui/_marker_gui.py:12: in <module>
    from mayavi.core.ui.mayavi_scene import MayaviScene
E   ImportError: No module named mayavi.core.ui.mayavi_scene
______________________________________________________________________ ERROR collecting mne/gui/_viewer.py _______________________________________________________________________
mne/gui/_viewer.py:9: in <module>
    from mayavi.mlab import pipeline, text3d
E   ImportError: No module named mayavi.mlab
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 5 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
=============================================================================== 2 tests deselected ===============================================================================
===================================================================== 2 deselected, 5 error in 3.98 seconds ======================================================================
Makefile:41: recipe for target 'test' failed
make: *** [test] Error 2
```

I don't have Mayavi installed and never had this problem before. Is Mayavi now necessary for mne-python? The last time I tried installing Mayavi, I ran into all sorts of problems wit my qt libraries and finally just decided to do without., so am hoping I don't need to refight that battle.

For the sake of completeness, here is my mne.sys_info():

```
>>> mne.sys_info()
Platform:      Linux-4.4.0-93-generic-x86_64-with-Ubuntu-16.04-xenial
Python:        2.7.12 (default, Nov 19 2016, 06:48:10)  [GCC 5.4.0 20160609]
Executable:    /usr/bin/python
CPU:           x86_64: 12 cores
Memory:        62.8 GB

mne:           0.15.dev0
numpy:         1.12.0 {lapack=openblas, blas=openblas}
scipy:         0.17.1
matplotlib:    2.0.0b3+1814.g3cb79ed

sklearn:       0.18
nibabel:       2.1.0
mayavi:        Not found
pycuda:        2016.1.2
skcuda:        0.5.2
pandas:        0.18.1
```
>>>

larsoner, 2017-09-16:
It shouldn't be required, this is a testing framework problem. Such imporn errors should be ignored or suppressed. Can you `git checkout cf9ff8bcbcb4fec21e8ebbef46305e60e6ead20d`? That tells us whether my test update PR or (probably) @massich's `py.test` transition PR broke things.

Either way I can probably fix it quickly on Monday.

smgutstein, 2017-09-18:
That checkout seems to work. The only error was:

E       IOError: [Errno 2] No such file or directory:
u'/home/smgutstein/mne_data/MNE-testing-data/KIT/test_as-raw.con'

which is easily fixed at my end.

PR #4576 (2017-09-20): MRG: Fix setup.cfg
Closes #4572.

@smgutstein can you verify that `py.test mne/gui` works for you now?
closed 2017-09-20 by agramfort