mne-tools/mne-python #4573: ENH/IO: add support for EGI Netstation EEG data - segmented files

christian-oreilly, 2017-09-17:
This ticket is closely related to the closed ticket  [`ENH/IO: add support for EGI Netstation EEG data. #1210`](https://github.com/mne-tools/mne-python/issues/1210). Issues 1210 implement a reader for MFF files, but only for unsegmented files. This ticket is to request support for segmented files. Inspiration can be taken from fieldtrip https://github.com/fieldtrip/fieldtrip/blob/master/fileio/ft_read_data.m which tackle with this issue in Matlab.

agramfort, 2017-09-17:
@christian-oreilly we'll see what we can do but don't hesitate to give it a try.

First can you share a file to test? the smaller the better?

christian-oreilly, 2017-09-17:
@agramfort This is actually a complicated request. I am stuck with 30 GB files but without NetStation to read them so that is why I try to use MNE (actually, Sleep which use MNE as a backend). I'll check in the days to come if I can get some collaborators to record with a NetStation system a few second recording.

However, trying to make some progess on this issue, I start looking at the code. From http://martinos.org/mne/dev/manual/io.html, we get that MFF files should be read using mne.io.read_raw_egi(). Is this an accurate information? I am asking because:

1. The simple binary files form EGI are also said to be read with the same function, although it is a different file format.
2. Because use big-endians, and from what I can see from fieldtrip and phypno, both these softwares use little endian when trying to read MFF files.
3. From what I can see from read_raw_egi, it seems to read the header following the structure of table 4-4 from [Net Station File Formats Technical Manual](https://www.google.ch/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwizroectKzWAhUBsRQKHSX4DtEQFggoMAA&url=http%3A%2F%2Flabs.psychologie.uva.nl%2Fbinaries%2Fcontent%2Fassets%2Fsubsites%2Ftechnische-ondersteuning-psychologie%2Fdownloads%2Fegi%2Fnet_station_file_formats.pdf%3F1376576303805&usg=AFQjCNE_USa8tZ0wUVZ7E4ILntSk4yS6fw) but I am not sure this is the right information for MFF format. In this manual, the first field should be the version and can take values 2, 3, 4, 5, 6, 7. When I read my own MFF files, the value of the first field (using small-endian) is 1 (using big endian is a very large value).

So, I was wondering if the MFF reader of MNE is indeed supposed to be read_raw_egi or if this is just a mistake in the documentation and this MFF reader is some other MNE function?

agramfort, 2017-09-17:
you point to the dev documentation. Did you try with current master code?

maybe someone with experience using these data will pitch in.... cc
@teonbrooks @ramonapariciog @dengemann @LauriParkkonen ?

christian-oreilly, 2017-09-17:
Sorry, silly mistake from my part. You were quite right in pointing that I should be running the dev version. mne was pip-installed using the latest release, not the dev version. I'll make a few checks to ensure that it is running smoothly and I will either close the ticket if it all works fine with the dev version or post an update about any remaining issue with using the dev version...

christian-oreilly, 2017-09-17:
Closing this issue. There was a few bug to correct (apparently, the function has not been tested with recording having no events and such recordings was causing the reader to crash at different places) but nothing related to the initial topic mentioned in this issue. I will submit a pull request with the bug correction.

closed 2017-09-17 by christian-oreilly