mne-tools/mne-python #4575: No amplitude from evoked.get_peak()

jjnurminen, 2017-09-18:
Wouldn't it be logical for `evoked.get_peak()` to also return the max amplitude? 
I coded this myself into a script that uses `get_peak()`, but I had to:

-arrange a new data matrix and channel names according to the pair grads logic
-find the index of the channel with maximum amplitude

`get_peak()` would already do these for free. However returning the amplitude would break API, as a 3-tuple of outputs would be needed.

larsoner, 2017-09-18:
We can add a return_amplitude to avoid API breakage

jjnurminen, 2017-09-18:
Yep. Then that would also be needed for the underlying machinery of `_get_peak()`.

larsoner, 2017-09-18:
`_get_peak` is private you can modify the API without adding an argument

jjnurminen, 2017-09-18:
Ahh, of course.

larsoner, 2017-09-18:
But in any case, +1 from me for this ENH

agramfort, 2017-09-18:
no objection

jjnurminen, 2017-09-19:
Ok, I will try to cook a relevant PR.

PR #4718 (2017-10-31): [MRG] return peak amp in get_peak()
Closes #4575
closed 2017-10-31 by larsoner