mne-tools/mne-python #4577: UserWarning in decoding_rsa example

arokem, 2017-09-19:
Running this example, I get the following warning 
    decoding_rsa.py:94: RuntimeWarning: More events than colors available. You should pass a list of unique colors.

jona-sassenhagen, 2017-09-19:
That's ok I would say. It comes from `plot_events`, and it will happen in many designs because there simply aren't that many distinguishable colours. It's only a warning after all.

Reopen if you disagree.

larsoner, 2017-09-19:
We should avoid warnings in examples wherever possible. I'd rather we fix
it by doing whatever we would like our users to do in this situation.

kingjr, 2017-09-19:
remove plot_events, it's unnecessary?

larsoner, 2017-09-19:
@kingjr can you make a quick PR?

ARudiuk, 2017-10-15:
Hello, this still seems to be an issue, and I was wondering if I could help? If the fix is to remove plot_events I can do that in a quick pull request.

larsoner, 2017-10-15:
+1 for removing it

agramfort, 2017-10-15:
go for it

PR #4664 (2017-10-17): [MERGE] Removed plot from decoding_rsa example
Removed plot that brought up warnings in decoding_rsa example.

closes #4577
closed 2017-10-17 by agramfort