mne-tools/mne-python #4580: ENH/IO: add support for EGI Netstation MFF file including physiological signals.

fraimondo, 2017-09-19:
Indeed I have to read several MFF including physiological signals (PSG).

The current MFF reader parses everything correctly, but only the EEG. This is due to the fact that PSG comes in a second file (signal2.bin) inside the MFF bundle.

I am ready to add support for it, but I wanted to discuss the approach first.

The `_read_mff_header` part is simple, since it need to check if the PSG data is there, then check the `pnsSet.xml` file to pull the names, and create the correct info.

The problem comes with `_read_segment_file`. It has to read some data from `signal1.bin` and the PSG data from `signal2.bin`.

There is a `self._filenames` array and a `fi` parameters, but they are supposed to be used for chunks in the time domain, and not channel wise.  Am I correct here?

So, to summarise, the constructor has to keep track of all the files that contains data, and the `_read_segment_file` should read both to construct the data for that segment.

So far the options:
* Use the `_filenames` attribute
* Create a new attribute containing the names of the PSG data files.
* Asume that if there is a `signal1.bin` and PSG channels in the info, there is a `signal2.bin` next to it.
 
What do you suggest/recommend?

agramfort, 2017-09-19:
see how edf reader does it with custom info keys. Whatever works but avoid
touching common info object.

fraimondo, 2017-09-19:
The `RawMFF` object already has a `self._filenames = [file_bin]` attribute. I would add a `self._psgfilenames` attribute to keep the info on where to read the PSG data. The rest is in the info dict as it is only channel names/types.

agramfort, 2017-09-19:
sounds ok for me.

fraimondo, 2017-09-19:
Need the EGI file format specification. Something's wrong with the current code regarding sampling rate. It's half of what it is supposed to be. Although for the testing data is fine. 

Anyone has the technical doc?

larsoner, 2017-09-19:
Don't do it that way, it won't work with concatenation. Store anything
extra you need in the _raw_extras list (usually a list of one dict per
"file", which in this case is per two-file set)

fraimondo, 2017-09-19:
I started doing that as soon as I realised that the raw extras can contain the name of the PSG datafile and channels info, if there was one.

fraimondo, 2017-09-20:
Now I've got a 20 sec test recording with PSG data, also exported into Matlab format to test the reader. Maybe you want to add it to the MNE testing dataset.

It's [here](https://www.dropbox.com/sh/z6shib5s6t6q9h8/AAA-STQvxZYzbJVEbU6cMOnYa?dl=0)

fraimondo, 2017-09-20:
So far the current reader has a difference of two samples between the Matlab (exported by net station and read by scipy) and the MFF with the MNE reader.

This is an EEG channel (so it's not related to the PSG data). 
[image]

I'm working to isolate and correct the issue.

agramfort, 2017-09-20:
ok. Let us know when we should have a look. thanks a lot

fraimondo, 2017-09-20:
Just finished.

So the `_read_segment_file` is mostly rewritten.

With the current code, there is one restriction that applies: if the indexes of the channels to be read are interlacing EEG and PSG data, then it raises an error.

There are alternatives to fix that problem, but it needs to be rewritten in a way that both EEG and PSG signal files are read simultaneously.

There might be some bugs if the API is forced to the limits. But there is a simple workaround: preload and pick after reading.

Let's follow the discussion in #4583, the new testing data should be added to the dataset before merging.

closed 2017-11-01 by fraimondo