mne-tools/mne-python #4586: Question: multiple runs => multiple coregistrations?

kingjr, 2017-09-19:
Imagine a subject is recorded multiple times, each with a different head positions the scanner, but with the very same coil positions on his/her face. 

Can we do a coregistration on the first run only, and then apply the inverse modeling for each run separately?

or 

Do we necessarily need to do one coregistration per run and save a different trans file? 

In the second case, shouldn't we be able to re-use the `info` from a single coregistration? 

cc @britta-wstnr  @larsoner

christianbrodbeck, 2017-09-19:
At least in our (KIT) workflow, the trans file stores the head-MEG transform. If you only digitize the head once per subject then there is only one head-MEG transform. What changes is the markercoils-head transform (assuming you acquire marker positions in every run). This is generated in the kit2fiff transform and stored in the FIFF file info, so you still need different inverse solutions.

kingjr, 2017-09-19:
Thanks @christianbrodbeck. That is an Elekta question though. But IIUC that means that a single coregistration is necessary right? I can just re-estimate (the covariance and) the inverse solution for each run separately, can't I.

christianbrodbeck, 2017-09-19:
I think that's correct. I am not familiar with Elekta, but the coregistration GUI shows (and create a trans file for) the digitized head shape. I assume head shape digitization is done only once regardless of system, and markercoil-head coregistration is run-dependent. To be sure you could check for one subject that the head shape coordinates are identical between FIFF files.

larsoner, 2017-09-19:
In principle you can, in practice you might be able to. Do you and your
digitizer record the fiducial locations without any meaningful error every
time you do a digitization?

In Elekta you generally digitize fiducials, plus at least a hundred extra
points. In principle, you only need the three fids as they establish the
head coordinate system, and this never changes. In practice, though, there
are errors in digitizing these points due to human and digitizer errors.
The hundred extra points (along with e.g. ICP fitting) ideally help reduce
this error by giving orders of magnitude more points to use in the fit.

You should be able to test this easily. Create N transforms for N
digitizations that might need unique head_mri_t's for one of your subjects,
and check your translation and rotation differences.

britta-wstnr, 2017-09-19:
But the HPI coil  initialization is done for every run, right? So if the participant moved, the MEG head space would be different, even with the same fiducial and headshape point positions.
Since MEG decive space and MRI space are the same across runs (within participant), and we obtain the MEG device -> MEG head space transform for every run, would it be possible to compute the transform to MRI space of runs 2 to 4 after coregistering run 1?

kingjr, 2017-09-19:
@christianbrodbeck, in the present case, the head shape is only acquired once, before all runs (all run are being recorded one after the other).

@larsoner I don't think I understand your answer. Overall, I share Britta's question

larsoner, 2017-09-20:
I misunderstood your question. I thought you were doing multiple recordings of the same subject e.g. on different days, and thus with different digitizations. If you don't redo any digitization and the coils don't move (a common assumption in back-to-back recordings), yes a single coreg will suffice.

britta-wstnr, 2017-09-20:
@larsoner thanks for clarifying this. What would happen now if we assume that the subject gets out of the sensor between runs (with fiducials and HPI coils staying in place) - this could change the coil positions relative to the sensors quite a lot (although they stay in place on the subject's head). 

I guess the question here is: is there a shortcut (and possibly an MNE-way to this shortcut) of coregistering every run individually with using the GUI only once (on run 1). 
E.g., by using the run-specific device -> MEG head space transform in ``info`` in combination with the trans file obtained for run 1 to calculate the transform to MRI space for runs 2 to x analytically.

larsoner, 2017-09-20:
There seems to be some confusion about terminology here and what gets used when.

When you digitize a subject, you get the fiducials, which define the head coordinate frame. Then all other points (EEG electrodes, extra points, HPI coils) are effectively registered as / transformed to points in this head coordinate frame.

> (with fiducials and HPI coils staying in place)

We know immediately based on this stipulation (whatever you say before or after does not affect it) is that the head<->MRI transform does not change. The head coordinate frame is defined by the digitized fiducials relative to the physical anatomy, which haven't changed. The MRI coordinate frame is defined by the MRI acquisition system (can't remember exactly how it's defined) but it clearly does not change. So the transformation between them does not change.

> What would happen now if we assume that the subject gets out of the sensor between runs ... - this could change the coil positions relative to the sensors quite a lot (although they stay in place on the subject's head).

This is independent of the head<->MRI transform. It will change the head<->MEG (a.k.a. head<->device or `dev_head_t` transform).

> by using the run-specific device -> MEG head space transform in info in combination with the trans file obtained for run 1 to calculate the transform to MRI space for runs 2 to x analytically.

I don't know what "MEG head space" is. The coordinate frames we are talking about here / MNE uses are head, MEG/device, and MRI. Can you rephrase it using these?

kingjr, 2017-09-21:
We solved that issue offline, but for future references:

* The confusion partially originates from the discrepency with Fieldtrip, which does source reconstruction in the MEG device space.

* The HPI and transform procedure is detailed in the comments of https://github.com/mne-tools/mne-python/blob/master/mne/chpi.py#L4

* Overall, we can apply a single coregistration and save a unique trans file, to be re-used by multiple successive runs. Each run (epochs/raw) stores the `dev_head_t` and the HPI position.

closed 2017-09-21 by kingjr