mne-tools/mne-python #4589: CTF: coregistration issue

kingjr, 2017-09-20:
With @avkasdan, we're having some issues with the coregistration of CTF data.

we used `mne.io.read_raw_ctf`  to read save the MEG data as a `fif` file.

we then use `mne.gui.coregistration` but the first issue is that we cannot save the trans file (the save button is not activated)
[image]

After digging down, it seems that changing the `info['dig']` of the fif file, solves this issue. Specifically, we changed all entries of `kind==3` into `kind=1` (no idea why, but this seems to be the sufficient thing to change to make `save_as` reappear).

However, now we still are unable to fit the fiducials with the HPI, even with MRI rescaling:
[image]

Any help would be greatly appreciated: the freesurfer, fif, and ctf (`*.ds/`) data is available here:
https://www.dropbox.com/sh/tqiymp8ymo62n85/AAAl1D0WKY3efMR-UeNH3qoOa?dl=0

(Linked to #3243?)

kingjr, 2017-09-20:
BTW, there is no issue with mne_analyze

larsoner, 2017-09-21:
@kingjr does this issue persist if you use MNE-C to convert to FIF instead of Python? This will tell us if the problem is with our CTF reader, or with `mne.coreg`.

@christianbrodbeck do you have time to look?

avkasdan, 2017-09-21:
Yes, the issue persists- still cannot save the trans file when the FIF file is created using MNE-C..

christianbrodbeck, 2017-09-21:
It looks to me like your raw file has missing digitizer information:

```
In [4]: raw.info['dig']
Out[4]: 
[{'coord_frame': 4,
  'ident': 3,
  'kind': 2,
  'r': array([ -4.33680869e-19,   1.05410405e-01,   0.00000000e+00], dtype=float32)},
 {'coord_frame': 4,
  'ident': 1,
  'kind': 2,
  'r': array([ -6.97835535e-02,   1.18220614e-19,   0.00000000e+00], dtype=float32)},
 {'coord_frame': 4,
  'ident': 2,
  'kind': 2,
  'r': array([  7.53390118e-02,  -6.40293176e-20,   0.00000000e+00], dtype=float32)},
 {'coord_frame': 4,
  'ident': 1,
  'kind': 3,
  'r': array([-0.02609644, -0.09746814,  0.0735818 ], dtype=float32)},
 {'coord_frame': 4,
  'ident': 2,
  'kind': 3,
  'r': array([ 0.03674514, -0.09581216,  0.0735818 ], dtype=float32)},
 {'coord_frame': 4,
  'ident': 3,
  'kind': 3,
  'r': array([ 0.07141331, -0.07693147,  0.0445022 ], dtype=float32)}]

```

Based on the `kind` entry, you have 3 EEG points (`FIFF.FIFFV_POINT_EEG = 3`) and 3 HPI points (`FIFF.FIFFV_POINT_HPI = 2`). The GUI needs at least the three cardinal points (Nasion, LPA, RPA) for an initial alignment. I don't think I ever tried using it without head shape points at all but without cardinal points you can't start in the current transformation implementation.

christianbrodbeck, 2017-09-21:
> However, now we still are unable to fit the fiducials with the HPI, even with MRI rescaling

From my GUI uniform scaling is enabled after pretending the HPI are cardinal points, just not 3-axes scaling. However from what I remember from playing around with it when I also had a head shape, scaling transforms based on just 3 points tended not to be very reliable...
, when I tried to estimate all parameters including scaling along 3 axes form just 3 points the results would net be very

larsoner, 2017-09-22:
> Based on the kind entry, you have 3 EEG points (FIFF.FIFFV_POINT_EEG = 3) and 3 HPI points (FIFF.FIFFV_POINT_HPI = 2). 

This is a case where we might want to add an option to (or by default) change what we do with CTF data. Those HPI coils are at the cardinal landmark locations (and might always be), so we could add e.g. `raw.add_fiducials_from_hpi()`. This would see if there are exactly three coils that are candidates for cardinal points (z=0, and then x=0 for one and y=0), and then would make duplicate entries for `info['dig']`.

This is actually orthogonal to #4593, so we can do it separately.

larsoner, 2017-09-22:
Another option is to make the coreg gui magically do this, but I'd rather have people be explicit, and we use cardinals elsewhere so it would be nice to *always* have them in `info['dig']`, so I don't like that option.

Yet another option is to always do this detection and add the HPI points as cardinal points in `read_raw_ctf`, but I'm not convinced this is as useful, as it doesn't fix the problem for people who have already converted their data to FIF, or use the C tools to do so.

christianbrodbeck, 2017-09-22:
Could we add a `Info.get_cardinal_point(s)` method that is sensitive to its contents?

larsoner, 2017-09-22:
Yes but it's not as useful. We have a lot of functions that use `info['dig']`, if we add entries to `info['dig']` all of these will just work (and probably should just work). If we make a method to get it, we need to modify all of them.

christianbrodbeck, 2017-09-22:
I know, but it will work with data that has already be converted... more work but better user experience? I don't have a strong feeling though as I don't use CTF :)

agramfort, 2017-09-23:
+1 to simply user experience

I would propose some magic in coreg unless the same magic needs to be
propagated
in other places

kingjr, 2017-09-24:
+1 to simply user experience too

agramfort, 2017-09-24:
let's make this happen this week so we can include it in the release

larsoner, 2017-09-24:
Okay if we don't want to make people do it explicitly we can do something like @christianbrodbeck suggested. I'll try this week

avkasdan, 2017-09-25:
We're a bit lost

What are cardinals?

And why does the coregistration work with `mne_analyze` but not `mne.gui.coregistration`.

JR

PR #4603 (2017-09-27): MRG: Fix CTF fiducials
Closes #4589.

@kingjr can you see if it fixes your CTF coreg problem?
closed 2017-09-27 by agramfort