mne-tools/mne-python #4606: BUG: sklearn cross_val_predict changed, now incompatible with SlidingEstimator with predict proba

kingjr, 2017-09-26:
sklearn `cross_val_predict` has recently changed, and is now incompatible with SlidingEstimator with classification estimators, as it requires `n_classes`

```python
import numpy as np
from mne.decoding import SlidingEstimator
from sklearn.linear_model import LogisticRegression, LinearRegression
from sklearn.model_selection import cross_val_predict
X = np.random.randn(100, 20, 10)
y = np.random.randint(0, 2, 100)

base_estimator = LinearRegression()
cross_val_predict(SlidingEstimator(base_estimator), X, y)  # ok

base_estimator = LogisticRegression()
cross_val_predict(SlidingEstimator(base_estimator), X, y)  # ok

base_estimator = LogisticRegression()
cross_val_predict(SlidingEstimator(base_estimator), X, y,
                             method='predict_proba')  # not ok
```

Three possible fixes:
1) Detect when `base_estimator ` is classifier, and add n_classes attribute to SlidingEstimator
2) make `mne.decoding.cross_val_multipredict`, similar to `mne.decoding.cross_val_multiscore`
3) add a 'method' argument to SlidingEstimator

agramfort, 2017-09-26:
don't we use cross_val_predict only in examples? if so why not passing
n_classes in our examples code?

kingjr, 2017-09-26:
I don't understand. 

It's more general issue of making SlidingEstimator compatible with sklearn, not just an example issue.

agramfort, 2017-09-26:
what do you want to do? what code breaks and should not?

kingjr, 2017-09-26:
? Please read above description, and possible fixes.

agramfort, 2017-09-26:
you need to add this:

https://github.com/scikit-learn/scikit-learn/blob/master/sklearn/model_selection/_search.py#L553

to our metaestimators

kingjr, 2017-09-27:
Ok I tried, but it's more complicated than this, because now, sklearn `cross_val_predict` initialize the shape of `y_pred` array based on the number of samples and the number of classes. However, with `SlidingEstimator`, `y_pred` is of shape `(n_samples, n_times, n_classes)`.

This leaves us with options 2 or 3.

agramfort, 2017-09-27:
then we need cross_val_multipredict ...

kingjr, 2017-09-27:
I think I join you on the '...' 

Shall I try to get this for the release, or is it better to give us some time to think?

The quick fix for now is to do e.g.

```python
class MyClf(LogisticRegression):
    def predict(self, X):
         return self.predict_proba(X)

cross_val_predict(MyClf(), X, y))
```

jona-sassenhagen, 2017-10-06:
More generally, I think `cross_val_multipredict` would be useful. (Not volunteering :D)

agramfort, 2018-08-04:
see #5394

PR #5394 (2018-08-06): FIX : cross_val_predict with SlidingEstimator using predict_proba
closes #4606
closed 2018-08-06 by larsoner