mne-tools/mne-python #4613: Using CHB-MIT seizure dataset in EDF format

syedumaramin, 2017-09-27:
I was wondering how could I use MNE for CHB-MIT seizure dataset classification which does not have stim channel and hence I can not find events.

agramfort, 2017-09-27:
please share a file or tell us what you typed to try

syedumaramin, 2017-09-27:
the CHB-MIT eeg database has no annotation channel so how can I use MNE to find events like seizure. 
It gives error that " no STIM channel found

agramfort, 2017-09-27:
how are stored the annotations? in the edf file?

please give me an example file

syedumaramin, 2017-09-27:
Here is the link for the files. These are edf files and seperate annotation file but no annotation channel
https://physionet.org/physiobank/database/chbmit/

agramfort, 2017-09-27:
I have no problem reading a file with current master:

raw = mne.io.read_raw_edf('chb01/chb01_01.edf', stim_channel='auto')

the annotation of seizures are stored outside of the edf AFAIK see
eg chb01_03.edf.seizures

don't bother about stim channel

syedumaramin, 2017-09-27:
thanks but how can i find events for classification purpose

syedumaramin, 2017-09-27:
Ho can I find events in a file with no annotation channel

agramfort, 2017-09-27:
you need to read the seizure file and do the maths to know where it
corresponds to in the file. this is not an MNE problem.

closed 2017-09-27 by agramfort

syedumaramin, 2017-09-27:
@agramfort , thanks

nurinathirahh, 2021-03-11:
hi @syedumaramin can you share how did you manage to read the .seizure file? i am a beginner in this EEG im having difficulty to read the .seizure using mne. Thank you so much

saiful1604008, 2022-05-29:
Can anyone share the pre-processing code of CHB-MIT dataset. I am new in EEG and facing difficulty to read the files. My email - u1604008@student.cuet.ac.bd

Thank you

drammock, 2022-05-29:
@saiful1604008 please use the MNE forum for usage questions. https://mne.discourse.group