mne-tools/mne-python #4614: Amplitude scale

MichlF, 2017-09-27:
Hi all,

It was suggested to me to post my feature request/suggestion here instead of via the mailing list, so here it goes:
There is currently no indication of the amplitude scale when inspecting individual epochs. I am doing visual research and I sometimes use the VEOG and HEOG channels to identify eye movements. Oftentimes there is a clear relationship between e.g. the size of the eye movement and the increase/decrease in amplitude of the step function. So for situations like these, and others, it would be handy to have an indicator of the scale (for an example of implementation in EEGLab see bottom right in this figure http://cognitrn.psych.indiana.edu/busey/temp/eeglabtutorial4.301/rejtut/rawpink1.gif ).

Thanks.

agramfort, 2017-09-27:
I think it's a useful feature indeed

we would need to have such scale bars for each visible channel type

larsoner, 2019-08-21:
FYI we have this for raw plotting, we need to port it to Epochs. Ideally we would also unify our Epochs and raw plotting a bit before then.

larsoner, 2021-02-19:
Porting the scale bars should have automagically happened when @drammock unified the raw and epochs browsers

closed 2021-02-19 by larsoner