mne-tools/mne-python #4615: "ImportError: No module named parallel" when launching mne

claire-braboszcz, 2017-09-27:
Step to reproduce (`python 2.7.12`):

- install `mne` from git using [these instructions](http://martinos.org/mne/dev/install_mne_python.html#install-the-mne-module) (`python setup.py develop`)
- run `mne`

Fix:

In `mne/decoding/search_light.py`, replace `from .parallel import parallel_func` by `from ..parallel import parallel_func`

(since `parallel.py` is not in the same directory, I wonder why it was working for me until yesterday...)

agramfort, 2017-09-27:
there is 2 "." in our code:

https://github.com/mne-tools/mne-python/blob/master/mne/decoding/search_light.py#L9

claire-braboszcz, 2017-09-27:
that's weird as I installed from scratch... anyway, good to know

agramfort, 2017-09-27:
how did you install?

closed 2017-09-27 by agramfort