mne-tools/mne-python #4631: find_ch_connectivity doesn't find EEG templates

cbrnr, 2017-10-02:
It looks like `mne.channels.find_ch_connectivity` only searches for MEG template files. This means that it doesn't work for EEG montages such as e.g. "biosemi64", even though a template exists. I think a note should be added in the docstring to clarify this behavior. Also, the note in `mne.channels.read_ch_connectivity` should be changed. Ideally, a good separation of the two functions is that `read_ch_connectivity` reads from template files whereas `find_ch_connectivity` computes connectivity.

agramfort, 2017-10-02:
PR welcome

cbrnr, 2017-10-02:
OK - I guess clarifying the docstrings is sufficient (the other alternative would be to add support for automatic detection of EEG montages).

agramfort, 2017-10-02:
if you think the automatic way can have a low enough failure mode sure why
not.

PR #4636 (2017-10-03): [MRG] Clarify the docstrings of read_ch_connectivity and find_ch_connectivity
Fix #4631. Let me check if the function links in the HTML docs work.
closed 2017-10-03 by agramfort