mne-tools/mne-python #4642: Units of PSD wrong?

cbrnr, 2017-10-04:
`mne.raw.plot_psd` labels the PSD with either "uV/sqrt(Hz)" (if `dB=False`) or "uV/Hz (dB)". Both seem to be incorrect as they should be "uV**2/Hz" and "dB", respectively.

larsoner, 2017-10-04:
The former is correct as we do the square root:

https://github.com/mne-tools/mne-python/blob/master/mne/viz/raw.py#L582

The latter is a bit loose in the terminology. I don't mind saying `uV**2/Hz (dB)`, because we are showing data that (formerly) had those units that has been converted to dB, but if purity (or possibly clarity?) is important we can go with just `dB`.

cbrnr, 2017-10-04:
> The former is correct as we do the square root

Why? Technically this is an ASD.

Regarding the decibel issue, yes, you could even add a suffix such as e.g. dBV, but this is not SI compatible. The current label if fine although I'd slightly prefer the pure dB.

Just to make sure, are you saying that `dB=False` produces an ASD whereas `dB=True` produces a PSD?

cbrnr, 2017-10-04:
Then I vote for changing that to a PSD (i.e. dropping the square root) - unless there is a good reason for having the ASD.

larsoner, 2017-10-04:
> unless there is a good reason for having the ASD.

The noise plots in linear units for MEG are often shown in these units

cbrnr, 2017-10-04:
The docs only mention PSD. Let's either update the docs or add an option to really compute a PSD or ASD (irrespective of the choice of `dB`).

larsoner, 2017-10-04:
PR welcome

aweinstein, 2017-10-04:
I can give the PR a try. Just to be clear, the right thing to do is to have an extra parameter to choose between PSD an ASD, and and let dB just set the units? Something like

```
dB : bool
    If True, transform data to decibels. If False, keep linear units.
scaling : { ‘density’, ‘spectrum’ }
    Selects between computing the power spectral density (‘density’) and the 
    amplitude spectral density ('spectrum').
```

aweinstein, 2017-10-04:
OK, although I still find confusing mixing the concepts of db/linear with PSD/ASD.

larsoner, 2017-10-04:
Perhaps just a doc update is sufficient?
```
dB : bool
    If True, show power spectral density estimates (power/Hz) transformed to decibels.
    If False, show amplitude spectral density estimates (amplitude/sqrt(Hz)).
```

cbrnr, 2017-10-04:
But I'd like to be able to plot a PSD with units uV**2/Hz too.

larsoner, 2017-10-04:
Okay so we do need the new parameter then. @aweinstein we'll have to live with the mixing for backward compatibility purposes.

aweinstein, 2017-10-09:
I need a little help with the PR. I'm just trying to add the `estimate` parameter to the `plot_raw_psd` function, and I get an exception from the `verbose` function. I changed the function call inside the `raw` 
class call method `plot_psd`, but I guess I need to change something else. The following is a diff of what I did, and the exception I get:

```
diff --git a/mne/io/base.py b/mne/io/base.py
index ed086ae..fac9879 100644
--- a/mne/io/base.py
+++ b/mne/io/base.py
@@ -1769,14 +1769,16 @@ class BaseRaw(ProjMixin, ContainsMixin, UpdateChannelsMixin,
                  color='black', area_mode='std', area_alpha=0.33,
                  n_overlap=0, dB=True, average=None, show=True,
                  n_jobs=1, line_alpha=None, spatial_colors=None,
-                 xscale='linear', reject_by_annotation=True, verbose=None):
+                 xscale='linear', reject_by_annotation=True, verbose=None,
+                 estimate='auto'):
         return plot_raw_psd(
             self, tmin=tmin, tmax=tmax, fmin=fmin, fmax=fmax, proj=proj,
             n_fft=n_fft, picks=picks, ax=ax, color=color, area_mode=area_mode,
             area_alpha=area_alpha, n_overlap=n_overlap, dB=dB, average=average,
             show=show, n_jobs=n_jobs, line_alpha=line_alpha,
             spatial_colors=spatial_colors, xscale=xscale,
-            reject_by_annotation=reject_by_annotation)
+            reject_by_annotation=reject_by_annotation,
+            estimate=estimate)
 
     @copy_function_doc_to_method_doc(plot_raw_psd_topo)
     def plot_psd_topo(self, tmin=0., tmax=None, fmin=0, fmax=100, proj=False,
diff --git a/mne/viz/raw.py b/mne/viz/raw.py
index 2a7c65e..c718a6d 100644
--- a/mne/viz/raw.py
+++ b/mne/viz/raw.py
@@ -566,8 +566,18 @@ def _set_psd_plot_params(info, proj, picks, ax, area_mode):
             ax_list, make_label)
 
 
-def _convert_psds(psds, dB, scaling, unit, ch_names):
-    """Convert PSDs to dB (if necessary) and appropriate units."""
+def _convert_psds(psds, dB, scaling, unit, ch_names, estimate=None):
     if dB:
         where = np.where(psds.min(1) <= 0)[0]
         if len(where) > 0:
@@ -588,10 +598,13 @@ def _convert_psds(psds, dB, scaling, unit, ch_names):
 @verbose
 def plot_raw_psd(raw, tmin=0., tmax=np.inf, fmin=0, fmax=np.inf, proj=False,
                  n_fft=None, picks=None, ax=None, color='black',
-                 area_mode='std', area_alpha=0.33, n_overlap=0, dB=True,
+                 area_mode='std', area_alpha=0.33, n_overlap=0,
+                 dB=True,
+                 #estimate='auto',
                  average=None, show=True, n_jobs=1, line_alpha=None,
                  spatial_colors=None, xscale='linear',
-                 reject_by_annotation=True, verbose=None):
+                 reject_by_annotation=True, verbose=None,
+                 estimate='auto'):
     """Plot the power spectral density across channels.
 
     Different channel types are drawn in sub-plots. When the data has been
@@ -700,8 +713,9 @@ def plot_raw_psd(raw, tmin=0., tmax=np.inf, fmin=0, fmax=np.inf, proj=False,
                                 n_overlap=n_overlap, n_jobs=n_jobs,
                                 reject_by_annotation=reject_by_annotation)
 
-        ylabel = _convert_psds(psds, dB, scalings_list[ii], units_list[ii],
-                               [raw.ch_names[pi] for pi in picks])
+        ylabel = _convert_psds(psds, dB, scalings_list[ii],
+                               units_list[ii],
+                               [raw.ch_names[pi] for pi in picks], estimate)
 
         if average:
             psd_mean = np.mean(psds, axis=0)

```

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
/home/ajw/Dropbox/tmp/mne_fix_psd/fix_psd.py in <module>()
     34 
     35     plt.close('all')
---> 36     raw.plot_psd(n_fft=1024, n_overlap=512, dB=True)

/home/ajw/.local/mne-python/mne/io/base.pyc in plot_psd(self, tmin, tmax, fmin, fmax, proj, n_fft, picks, ax, color, area_mode, area_alpha, n_overlap, dB, average, show, n_jobs, line_alpha, spatial_colors, xscale, reject_by_annotation, verbose, estimate)

/home/ajw/.local/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    726         with use_log_level(verbose_level):
    727             return function(*args, **kwargs)
--> 728     return function(*args, **kwargs)
    729 
    730 

TypeError: plot_psd() takes at most 22 arguments (23 given)
```

larsoner, 2017-10-09:
That's confusing, the code actually looks okay. FWIW in the meantime we always have `verbose` last by convention, so `estimate` should go before it.

aweinstein, 2017-10-09:
So I deleted `utils.pyc`, and now it works. I'm using the autoreload IPython magic. Perhaps is related to that.

larsoner, 2017-10-09:
FYI you can do `make clean` from the `mne-python` root dir to clean up all of these files

aweinstein, 2017-10-09:
Right now we throw a `ValueError` exception when there is a PSD value smaller or equal than 0, *only* if `dB=True`. Should't we throw that exception regardless of the units?

This is the code:

```
    if dB:
        where = np.where(psds.min(1) <= 0)[0]
        if len(where) > 0:
            raise ValueError("Infinite value in PSD for channel(s) %s. "
                             "These channels might be dead." %
                             ', '.join(ch_names[ii] for ii in where))
        psds *= scaling * scaling
        np.log10(psds, out=psds)
        psds *= 10
        ylabel = '%s/Hz (dB)' % unit
    else:
        np.sqrt(psds, out=psds)
        psds *= scaling
        ylabel = '$\\frac{%s}{\\sqrt{Hz}}$' % unit
```

cbrnr, 2017-10-19:
Fixed in #4654.

closed 2017-10-19 by cbrnr