mne-tools/mne-python #4643: Permutation cluster tests should compute vertex connectivity explicitly

wmvanvliet, 2017-10-04:
This has bitten two of my coworker in the ass. The permutation clustering examples on source space (e.g. https://martinos.org/mne/stable/auto_tutorials/plot_stats_cluster_spatio_temporal.html) use this to compute the connectivity between vertices:

    connectivity = spatial_tris_connectivity(grade_to_tris(5))

However, this is not always appropriate, depending on mixtures of old freesurfer brains and old mne-python versions. If the connectivity matrix is wrong, the statistics will go haywire in unpredictable ways. Since our users are prone to copy-pasting our example code without a second thought, this will be safer:

    # Read the source space that the individual stcs were morphed to
    src = mne.read_source_space('/path/to/fsaverage-ico4-src.fif')
    connectivity =  spatial_src_connectivity(src)

larsoner, 2017-10-04:
When we do this we'll also want to fix `spatial_src_connectivity` to work for `oct` source spaces. It seems like both should just use `spatial_tris_connectivity` under the hood, not sure whether or not they do currently.

wmvanvliet, 2017-10-04:
`spatial_src_connectivity` calls `spatial_tris_connectivity` under the hood, yes.

PR #4973 (2018-02-27): MRG: Better tutorials using connectivity
Closes #4643.

@wmvanvliet feel free to have a look, this should do what we talked about earlier to get rid of `grade_to_tris`.
closed 2018-02-27 by larsoner