mne-tools/mne-python #4647: "NO_DATA" when epoching data after crop, save, load

bdyetton, 2017-10-05:
Hi MNE Team,
It seems, that per[ this discussion,](https://mail.nmr.mgh.harvard.edu/pipermail/mne_analysis/2016-April/003249.html) that epoching raw data that has previously been cropped, saved and loaded again will cause any epochs outside the previous crop area to be dropped due to "NO_DATA".

I've just run into this problem myself and had to apply the ```raw._first_samps[0] = 0``` fix as suggested by Tristan in the discussion.

This seems like a bug to me, so I'm flagging it here.

larsoner, 2017-10-05:
Can you isolate / reproduce the problem with the sample dataset (`mne.datasets.sample.data_path()`)?

larsoner, 2017-10-05:
(And if not, can you reproduce it with a dataset of your own that you can share?)

bdyetton, 2017-10-05:
[Here](https://drive.google.com/open?id=0BzCaZAqMQTL6cEpERmlMNVJaT0k) is the data that is having the issue. 

```python
    import numpy as np
    import mne
    raw = mne.io.read_raw_fif('./test_raw.fif', verbose=True)
    raw.load_data()
    epochs = mne.Epochs(raw, np.array([[0, 0,0], [200,0,0]]), 0)
    epochs.drop_bad()
    print(epochs.drop_log)
```

larsoner, 2017-10-05:
This behavior seems correct to me. You are putting events at sample [0] and [200] and the first sample in your raw data (`raw.first_samp`) is 40890.

larsoner, 2017-10-05:
> Also, i just noticed that if i pick one channel only, or click on a channel in the plotted version of the data to delete it, then the other channels data goes to zero. Is this expected behavior?

An average projector with only one channel will zero out the channel. We should probably add some protection against this, though I'm surprised that you don't get a warning about the projector not containing enough of the original channels.

bdyetton, 2017-10-05:
So when cropping, the signal is not removed but instead, the ```first_samp``` is just sent to the start of the cropping? If so, that seems rather unintuitive behavior to me. "Cropping" in all other contexts (except images in MS Word) would remove the data outside of the crop area.

As for the channel going to zero, is it possible to apply the projector (such as an average reference) so that the raw data itself is modified? This way, if future channels are dropped, other channels will not change.

bdyetton, 2017-10-05:
Oh right, ignore that last question. The ```apply_proj()``` function will do that.

bdyetton, 2017-10-06:
Note that the ```apply_proj``` function does not seem to do this, as im still getting warnings about not enough channels and seeing zero'ed channels after applying projections.

agramfort, 2017-10-06:
I agree that cropping should not necessarily reset time axis

I am open to suggestions

larsoner, 2017-10-06:
What do you mean? In what sense does it reset the time axis?

agramfort, 2017-10-06:
I thought it was setting first_samp to 0 but it does not. So I also don't
understand the pb

larsoner, 2017-10-06:
> If so, that seems rather unintuitive behavior to me. "Cropping" in all other contexts (except images in MS Word) would remove the data outside of the crop area and reset the "zero" position.

The alternative (resetting the zero position) also results in unintuitive behaviors in this context, though. In the current design, if you crop data into, 3 contiguous segments, find events on each segment, and concatenate the resulting event lists, you get the same event list you would have gotten from the original (neglecting edge effects). The same goes in reverse -- if you find events on the original, and then crop your data to any window, epoching using those original events works.

Can you think of a way to update the docs to make this behavior clearer? Maybe something could be added to `raw.crop`.

> As for the channel going to zero, is it possible to apply the projector (such as an average reference) so that the raw data itself is modified? This way, if future channels are dropped, other channels will not change.

Yes, use `projection=False` in `set_eeg_reference`

bdyetton, 2017-10-06:
Hmmm, yea I see when dealing with events, it would be problematic. I guess there could be a parameter in the crop function that specifies if the axis should be reset or not. Maybe if this parameter is required it will force people to think about it, although that may break old code. 
And yes, at least some documentation to the crop function should be added!

dengemann, 2017-10-06:
Improving the docs is a good idea. I would not suggest to change the API.
The crop API has been there for ages and does its job well.

agramfort, 2017-10-07:
agreed

bdyetton, 2017-10-12:
Maybe not a change to the API, but the addition of an optional ```reset_first_sample``` or the like may be helpful. It seems I am not the only one who has hit this issue. However, i guess manually changing the ```raw._first_samps[0] = 0``` fixes the issue. Up to you guys :)

closed 2017-10-12 by bdyetton

larsoner, 2017-10-12:
The correct fix is to add `raw.first_samp` to `events[:, 0]`, as then they will work properly with your cropped instance or if you go back to your original data

larsoner, 2017-10-12:
(and will match events you get if you do `mne.find_events(raw_cropped, ...)` on your data, etc.)