mne-tools/mne-python #4648: Feature Request: Migrate MNE discussion to something like discourse

bdyetton, 2017-10-05:
Hi MNE Team,
I find the mailing list discussion of MNE issues a little outdated. 
I'm opening this feature request to migrate to something like [discourse ](https://www.discourse.org/). It makes finding relevant issues is much easier, and it seems to be better indexed by Google. 
I know you guys are super busy, and this should not be a high priority feature, but I thought I would log it here in case other users feel the same.
Thanks for all the great work on MNE so far!

agramfort, 2017-10-05:
if you have a way to sync a mailman list with discourse that would be neat

bdyetton, 2017-10-05:
Sounds like its [possible](https://meta.discourse.org/t/import-mailman-archives-into-discourse/18537), but not straightforward. It seems like a common request, and the discourse team looks like they are considering a simple sync process. It may be worth waiting until discourse fully features mailman integration.

drammock, 2020-11-17:
closed in favor of #8429 which has a thorough discussion of Discourse as well as other competing options.

closed 2020-11-17 by drammock