mne-tools/mne-python #4649: BUG: Plotting Cz fails

jona-sassenhagen, 2017-10-06:
With my current setup, plotting a sensor at 0, 0, 0 (e.g., Cz) fails because Numpy returns `False` for an array of all-zeros.

```
~/tools/mne-python/mne/viz/topomap.py in _prepare_topomap(pos, ax)
   1972     _hide_frame(ax)
   1973     if any([not pos_y.any(), not pos_x.any()]):
-> 1974         raise RuntimeError('No position information found, cannot compute '
   1975                            'geometries for topomap.')
   1976     return pos_x, pos_y

RuntimeError: No position information found, cannot compute geometries for topomap.
```

I'll try to take care of it.

larsoner, 2017-10-06:
We should probably use nan for undefined locations instead of zero, but
that's a hard change to make

In the meantime you can change the check to also look at the z coordinate
earlier in some calling function

jona-sassenhagen, 2017-10-10:
Uh ... what do you mean? The z coordinate is also 0 ...

I'm not sure how to deal with this, but I'd like to fix it quick, because plot_compare_evokeds with Cz is something I do quite frequently, and expect other ERP people to do frequently too.

larsoner, 2017-10-10:
> Uh ... what do you mean? The z coordinate is also 0 ...

In Neuromag head coordinates, Cz should not have a z coordinate of zero. It implies a completely flat (like a circle instead of a sphere) head

jona-sassenhagen, 2017-10-10:
Hm, ok - but these are x,y,z coordinates? Sorry, I don't really get what you're proposing :)

larsoner, 2017-10-10:
> these are x,y,z coordinates

Yes they are X/Y/Z but in what coordinate frame? In MNE we assume that EEG electrodes are in Neuromag head coordinates (defined by LPA at (-a, 0, 0), RPA at (b, 0, 0) and nasion at (0, c, 0)) at various places. Thus why an electrode at 0, 0, 0 and also on the surface of the head implies that the head is flat.

In other words:

1. you should try to get your coordinates into this coordinate frame (hopefully easy)
2. we should make the coordinate checker also check the z coordinate (also hopefully easy)
3. we should stop using (0, 0, 0) to mean "no position" and use `np.nan` (harder)

jona-sassenhagen, 2017-10-10:
Oh, ok. So the situation is this:

- the pos is extracted by `_auto_topomap_coords`
- it is then passed through `_check_outlines`
- the check happens inside `_plot_legend`

The problem is, `_auto_topomap_coords` already only returns x and y coordinates.

So when the coords are passed to `_prepare_topomap`, z is already missing.

`_prepare_topomap` does three things: check, split pos into pos_x and pos_y, and hide axes.

So I fear the z coordinate doesn't help us.

Thankfully, the problem is probably localised: it should only actually be a problem if there is only one channel, which probably only happens inside `plot_compare_evokeds`. It could be solved by any number of ways, but all I can think of are a bit hackish.

larsoner, 2017-10-10:
> _prepare_topomap does three things: check, split pos into pos_x and pos_y, and hide axes.

These are all private functions. How about we move the sanity check to `_auto_topomap_coords(..., check_positions=False)` by default with `check_positions=True` in this case, meaning "check to make sure there are are valid XYZ positions and raise an error if not". That way we catch the error before any conversion takes place, and `_prepare_topomap` no longer needs to do the check.

jona-sassenhagen, 2017-10-10:
In principle yes, though we probably want to undo the `check_positions` options once nan channels are truly nan.

However, at that stage, _prepare_topomap is seemingly only doing 2 things:
- hide the axes by calling _hide_axes
- converting `pos` into `pos[0, :], pos_y[1, :]`
So the function might look a bit unnecessary at this stage :D

jona-sassenhagen, 2017-10-10:
Either way, I'll hopefully make a PR like that soon.

PR #4688 (2017-10-20): Allow plotting Cz
See #4649

@larsoner , I've picked a hopefully less invasive way of going about it.

Still, eventually missing channel positions should become NaNs.

Closes #4649
closed 2017-10-20 by larsoner