mne-tools/mne-python #4650: Apply_inverse problem

hendriksantosa, 2017-10-06:
Hi All, 

Anyone can help me with the error when I run apply_inverse "EEG average reference is mandatory for inverse modeling, use set_eeg_reference" method. 
I don't have an EEG data. Also, I put EEG=False in the argument. 
It seems everything looks fine to me. Maybe anyone can help me to find the error.  

-----------------
Here is the script
-----------------
    src_morph = mne.morph_source_spaces(src,'fsaverage',subjects_dir=subj_dir)
    
    model = mne.make_bem_model(subject=subj, ico=4, conductivity=conductivity, subjects_dir=subj_dir)
            
    bem = mne.make_bem_solution(model)
 
    fwd_morph = mne.make_forward_solution(raw.info, trans=trans, src=src_morph, bem=bem, fname=None, meg=True, eeg=False, mindist=0.0, n_jobs=2)
    fwd_morph = mne.pick_types_forward(fwd_morph, meg=True, eeg=False)

    cov = mne.compute_raw_covariance(raw, tmin=0, tmax=30)
    
    inv_morph = mne.minimum_norm.make_inverse_operator(raw.info, fwd_morph, cov, loose=None, depth=0.8)

    stc_morph = mne.minimum_norm.apply_inverse_raw(raw, inv_morph, lambda2, method, pick_ori='normal', buffer_size=25000)

agramfort, 2017-10-06:
what error do you get? Please show us what print(raw.info) gives.

hendriksantosa, 2017-10-09:
Alex,

Here is the information:

print(raw.info)
<Info | 28 non-empty fields
    acq_pars : 'str | 21614 items
    bads : 'list | 0 items
    buffer_size_sec : 'numpy.float64 | 1.0
    ch_names : 'list | MEG0113, MEG0112, MEG0111, MEG0122, MEG0123, MEG0121, ...
    chs : 'list | 324 items (GRAD: 204, MAG: 102, STIM: 5, EEG: 8, EOG: 2, ECG: 1, MISC: 2)
    comps : 'list | 0 items
    custom_ref_applied : 'bool | False
    description : 'str | 17 items
    dev_head_t : 'mne.transforms.Transform | 3 items
    dig : 'list | 115 items
    events : 'list | 0 items
    experimenter : 'str | 3 items
    file_id : 'dict | 4 items
    highpass : 'float | 0.10000000149011612 Hz
    hpi_meas : 'list | 1 items
    hpi_results : 'list | 1 items
    hpi_subsystem : 'dict | 2 items
    line_freq : 'float | 50.0
    lowpass : 'float | 330.0 Hz
    meas_date : 'numpy.ndarray | 2017-09-18 15:04:54
    meas_id : 'dict | 4 items
    nchan : 'int | 324
    proc_history : 'list | 1 items
    proj_id : 'numpy.ndarray | 1 items
    proj_name : 'str | 8 items
    projs : 'list | 0 items
    sfreq : 'float | 1000.0 Hz
    subject_info : 'dict | 6 items
    acq_stim : 'NoneType
    ctf_head_t : 'NoneType
    dev_ctf_t : 'NoneType
    kit_system_id : 'NoneType
    xplotter_layout : 'NoneType

larsoner, 2017-10-09:
If you do `raw.pick_types(meg=True, eeg=False)` it should work as a workaround.

We must have some subtle problem where we check for avg eeg ref before all picking is done, as in this case the forward does not / should not have EEG channels.

hendriksantosa, 2017-10-09:
Thank you Eric, the issue has been resolved.

larsoner, 2017-10-09:
Good to hear @hendriksantosa. Let's leave this open because we do have a bug with our checking I think.

PR #5385 (2018-08-01): MRG: Fix check for avg ref
Closes #4650.
closed 2018-08-01 by agramfort