mne-tools/mne-python #4660: vector inverse for epochs data, plotting

emilyps14, 2017-10-12:
A couple of issues with running the tutorial plot_mne_dspm_source_localization with epoch data instead of evoked. Specifically, I am trying to compute a vector inverse using the line:
`stc_vec = mne.minimum_norm.apply_inverse_epochs(epochs[0], inverse_operator, lambda2, method=method, nave=1, pick_ori='vector', return_generator=False)[0]`

Upon further inspection I found some apparent bugs:
1. In line [1067](https://github.com/mne-tools/mne-python/blob/master/mne/minimum_norm/inverse.py#L1067) of inverse.py it compares `len(sel)` to `K.shape[0]` when in fact it should compare to `K.shape[1]`, the channel axis
2. In line [1042](https://github.com/mne-tools/mne-python/blob/master/mne/minimum_norm/inverse.py#L1042) of inverse.py it should not require that `pick_ori is None`, since later in line 1060 it checks for `pick_ori != 'vector'` (and behaves correctly when `pick_ori==vector`). 
3. If `_make_stc` is called with a tuple (which happens whenever line [1068](https://github.com/mne-tools/mne-python/blob/master/mne/minimum_norm/inverse.py#L1068) is executed), it breaks because it tries to access `data.shape`

I can compute the inverse with corrections (1) and/or (2), but I think someone more familiar with the intended logic should review it to make sure it makes sense.

larsoner, 2017-10-12:
The `pick_ori='vector'` is fairly new and AFAIK people have not used it with `apply_inverse_epochs` yet. Your fixes sound reasonable. Do you have time to make a pull request to fix it? We'd want to add some basic test that `apply_inverse_epochs` works -- since the `vector` is linear, we could test that the average of the epoch inverses is numerically equal to the `apply_inverse` on the average (assuming an MNE solution so there is no noise scaling).

emilyps14, 2017-10-12:
Sure, I'll take a shot at it.

closed 2017-10-17 by larsoner