mne-tools/mne-python #4663: FIX?: CTF ref meg channels not handled properly in LCMV

kingjr, 2017-10-13:
Reconstructing CTF (271 mag + ref meg) using LCMV seems buggy. If I don't explicitly exclude the ref meg, the reconstruction gets dramatically influenced by the ref channels, and lead to seemingly absurd estimates.

I think this is an issue with the way the forward model is used by lcmv.

@romquentin, didn't you have a similar issue at some point?

cc @britta-wstnr @avkasdan

To reproduce the issue: [data](https://wetransfer.com/downloads/a74715046dd742bfddc4c6620aa587bf20171013133120/02b00e7cd46948b32d393014aef5d2b420171013133120/561f29) and code:
```python
import mne
import numpy as np
import matplotlib.pyplot as plt
from mne.minimum_norm import apply_inverse_epochs, make_inverse_operator
from mne.cov import compute_covariance

# Setup paths and subject
subject = 'ESA28'

# Read in epochs and apply baseline
epochs = mne.read_epochs('%s_audio_None-30.0Hz-epo.fif' % subject)
epochs.apply_baseline((-1.000, -.900))

# Setup Forward model
bem_fname = '%s-5120-bem-sol.fif' % subject
src = mne.setup_volume_source_space(subject, pos=10, mri='T1.mgz',
                                    bem=bem_fname, subjects_dir='./')
fwd = mne.make_forward_solution(info=epochs.info, trans='%s_01-trans.fif' % subject,
                                src=src, ignore_ref=True, bem=bem_fname)

# Compute the noise and data covariance
# BUG don't exclude ref channel here, and you'll get crazy sources
epochs.pick_types(meg='mag', ref_meg=False)
noise_cov = compute_covariance(epochs, tmin=-1., tmax=-.900)
data_cov = compute_covariance(epochs, tmin=-.800)
epochs.decimate(3)

# Setup inverse operator
# FIXME loose='auto' doesn't work as it should
inv = make_inverse_operator(info=epochs.info, forward=fwd,
                            noise_cov=noise_cov, loose=1, depth=None)

stc = mne.beamformer.lcmv_epochs(epochs, fwd, noise_cov, data_cov,
                                 reg=0.05, pick_ori=None)

# Retrieve data to plot average
data = np.zeros_like(stc[0]._data)
for src in stc:
    data += src.data

plt.matshow(data)
```

The same code without ```epochs.pick_types(meg='mag', ref_meg=False)```python, seems to follow the ref channels.

romquentin, 2017-10-13:
I did not see this bug. I usually exclude ref_meg.

britta-wstnr, 2017-10-16:
This is indeed a bug within the ``lcmv`` code, thanks @kingjr . Working on the fix for it.

larsoner, 2018-04-04:
Should have been fixed by #4665, feel free to reopen if not

closed 2018-04-04 by larsoner