mne-tools/mne-python #4671: Exception when printin raw.info

aweinstein, 2017-10-16:
Printing the `info` attribute of a raw object, using 5fac33d, throws the following exception:

```
Traceback (most recent call last):
  File "eeg.py", line 361, in <module>
    print(raw.info)
  File "/home/ajw/.local/mne-python/mne/io/meas_info.py", line 410, in __repr__
    entr = _stamp_to_dt(v).strftime('%Y-%m-%d %H:%M:%S') + ' GMT'
  File "/home/ajw/.local/mne-python/mne/io/meas_info.py", line 64, in _stamp_to_dt
    datetime.timedelta(0, 0, stamp[1]))  # day, sec, μs
IndexError: list index out of range
```
I can print the `info` if I go back to 0c0b8de2. I think the commit that causes the problem is e6fd505ef4e35089ab0b31d69fe371028c79cd53 (but I can't go back to that commit).

This is the `info` value:

```
<Info | 19 non-empty fields
    bads : list | 0 items
    buffer_size_sec : float64 | 1.0
    ch_names : list | Fp1, AF7, AF3, F1, F3, F5, F7, FT7, FC5, ...
    chs : list | 65 items (EEG: 64, STIM: 1)
    comps : list | 0 items
    custom_ref_applied : bool | False
    dev_head_t : Transform | 3 items
    events : list | 0 items
    file_id : dict | 4 items
    highpass : float | 0.5 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 30.0 Hz
    meas_date : ndarray | 2015-02-04 08:21:37
    meas_id : dict | 4 items
    nchan : int | 65
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 150.0 Hz
    acq_pars : NoneType
    acq_stim : NoneType
    ctf_head_t : NoneType
    description : NoneType
    dev_ctf_t : NoneType
    dig : NoneType
    experimenter : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    proj_id : NoneType
    proj_name : NoneType
    subject_info : NoneType
    xplotter_layout : NoneType
>

In case it matters, the original data is from a BDF file.

larsoner, 2017-10-16:
Can you `import pdb; pdb.pm()` to find out what the `info` entry is that is causing the problem? It's probably some malformed timestamp that we can make a fix for.

aweinstein, 2017-10-17:
(It was easier to make a PR than to figure it out this pdb business). 

I think the problem is that BSD don't have microsecond information in the time of "Start time of recording" field (see [this](https://www.biosemi.com/faq/file_format.htm)).

PR #4672 seems to fix the problem.

agramfort, 2017-10-17:
can you share the file?

aweinstein, 2017-10-17:
> can you share the file?

https://www.dropbox.com/s/q9ba8mr3rbqgdqy/01EEG0410_ica_raw.fif?dl=0

PR #4672 (2017-10-17): Fix when there is no us information in mea_date date info value of raw object
Closes #4671
closed 2017-10-17 by larsoner