mne-tools/mne-python #4674: BUG?: multiple stim channels API seems broken or at least very unintuitive.

dengemann, 2017-10-17:
Try the following with the sample data:

```Python

mne.find_events(raw, stim_channel=['STI 014', 'STI 015'])  #-> no events
mne.find_events(raw, stim_channel=['STI 014'])  #-> all events
mne.find_events(raw, stim_channel=['STI 015', 'STI 014'])  #-> no events
mne.find_events(raw, stim_channel=['STI 015'])  #-> no events
mne.find_events(raw, stim_channel=['STI 001'])  #-> all events with value 5
mne.find_events(raw, stim_channel=['STI 001', 'STI 014'])  #-> all events with value 5
mne.find_events(raw, stim_channel=['STI 014', 'STI 001'])  #-> all events with value 5
```

Thoughts @agramfort @larsoner ?

agramfort, 2017-10-17:
seems like we don't catch bad param values in find_events

PR welcome

larsoner, 2017-10-17:
Yeah we should probably just get rid of support for list. I'm not even sure what the behavior should be, but in any case it's probably less explicit than using single channels and doing logic on the results.

dengemann, 2017-10-17:
> Yeah we should probably just get rid of support for list. I'm not even sure what the behavior should be, but in any case it's probably less explicit than using single channels and doing logic on the results.

The intuition that I had was something like an union over the channels. But this is not what it does.

jjnurminen, 2017-10-17:
Just out of curiosity, why would you feed both a combination channel (e.g. STI 014 or STI101) and an individual (single bit) trigger channel (e.g. STI 001) to find_events()? Just the former already contains all the trigger information.

agramfort, 2017-10-18:
see https://github.com/mne-tools/mne-python/pull/4678

closed 2017-10-18 by larsoner