mne-tools/mne-python #4680: RuntimeError when using read_epochs

Amyunimus, 2017-10-18:
I'm getting the following error when trying to read in an epochs file that I just created using `epochs.save('test-epo.fif.gz')`. I manually selected just the occipital MEG channels as `picks` for the epochs and BG1-4304 is not among them.  

It looks like this issue has come up before and has to do with a calibration channel (or lack thereof), but it was unclear how to fix it.

```
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-3-9d449d0556b7> in <module>()
----> 1 test =  mne.epochs.read_epochs('test-epo.fif.gz')

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/epochs.pyc in read_epochs(fname, proj, preload, verbose)

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/epochs.pyc in read_epochs(fname, proj, preload, verbose)
   2425         The epochs
   2426     """
-> 2427     return EpochsFIF(fname, proj, False, preload, verbose)
   2428 
   2429 

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/epochs.pyc in __init__(self, fname, proj, add_eeg_ref, preload, verbose)

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/epochs.pyc in __init__(self, fname, proj, add_eeg_ref, preload, verbose)
   2494             (info, data, data_tag, events, event_id, tmin, tmax, baseline,
   2495              name, selection, drop_log, epoch_shape, cals) = \
-> 2496                 _read_one_epoch_file(fid, tree, preload)
   2497             # here we ignore missing events, since users should already be
   2498             # aware of missing events if they have saved data that way

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/epochs.pyc in _read_one_epoch_file(f, tree, preload)
   2286     with f as fid:
   2287         #   Read the measurement info
-> 2288         info, meas = read_meas_info(fid, tree, clean_bads=True)
   2289 
   2290         events, mappings = _read_events_fif(fid, tree)

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/io/meas_info.pyc in read_meas_info(fid, tree, clean_bads, verbose)

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/io/meas_info.pyc in read_meas_info(fid, tree, clean_bads, verbose)
    775 
    776     #   Load the CTF compensation data
--> 777     comps = read_ctf_comp(fid, meas_info, chs)
    778 
    779     #   Load the bad channel list

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/io/ctf_comp.pyc in read_ctf_comp(fid, node, chs, verbose)

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/utils.pyc in verbose(function, *args, **kwargs)
    705         with use_log_level(verbose_level):
    706             return function(*args, **kwargs)
--> 707     return function(*args, **kwargs)
    708 
    709 

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/io/ctf_comp.pyc in read_ctf_comp(fid, node, chs, verbose)
    109         if not calibrated:
    110             #   Calibrate...
--> 111             _calibrate_comp(one, chs, mat['row_names'], mat['col_names'])
    112         else:
    113             one['rowcals'] = np.ones(mat['data'].shape[0], dtype=np.float)

/home/predatt/emiwar/local/lib/python2.7/site-packages/mne/io/ctf_comp.pyc in _calibrate_comp(comp, chs, row_names, col_names, mult_keys, flip)
     42             if p != 1:
     43                 raise RuntimeError('Channel %s does not appear exactly once '
---> 44                                    'in data' % names[ii])
     45             idx = ch_names.index(names[ii])
     46             val = chs[idx][mult_keys[0]] * chs[idx][mult_keys[1]]

RuntimeError: Channel BG1-4304 does not appear exactly once in data
```

larsoner, 2017-10-18:
Have you tried latest `master` or are you on 0.14?

Amyunimus, 2017-10-18:
I was using 0.14.1, but just now tried it on the latest `master` (0.15.dev0) and got the same thing.

agramfort, 2017-10-18:
please share file(s) and a script so we can replicate

Amyunimus, 2017-10-18:
These are the epochs (compressed pickle file, since I can't save it another way): https://www.dropbox.com/s/b9qb64ygodypeic/epochs.pklz?dl=0

And this code should replicate the problem:
```
import gzip
import pickle
import mne

f = gzip.open('epochs.pklz','rb')
epochs = pickle.load(f)
f.close()

print epochs.info

epochs.save('test-epo.fif.gz')
 
epochs_from_file =  mne.epochs.read_epochs('test-epo.fif.gz')
```
I can provide more code/raw file if needed, but the raw file is enormous, so not as easily sharable. It's CTF.

larsoner, 2017-10-18:
Can you post the lines of code you used for raw->picking->epoching? I suspect that with those we can probably reproduce with existing CTF datasets.

Amyunimus, 2017-10-18:
Here it is (w/o event definition)

```python
raw = mne.io.read_raw_ctf(raw_fname, preload=False)
picks = np.array(range(87, 106) + range(218, 237) +range(300, 303)) #occipital channels
reject = {'mag': 4e-12}
epochs = mne.Epochs(raw, events=events, tmin=-0.1, baseline=(None, 0.0),
                    tmax=5, picks=picks, preload=True, reject=reject)
```

Amyunimus, 2017-10-24:
Are there any workarounds for this? I'm running into some memory constraints and need to offload the epochs to disk.

agramfort, 2017-10-24:
please provide a full gist using the spm face ctf dataset we have.

http://martinos.org/mne/stable/auto_examples/datasets/plot_spm_faces_dataset.html

Amyunimus, 2017-10-24:
This replicates the ``RuntimeError: Channel BG1-2908 does not appear exactly once in data``  error with the spm face ctf dataset:

```python
import numpy as np
import mne
from mne.datasets import spm_face

data_path = spm_face.data_path()
subjects_dir = data_path + '/subjects'

raw_fname = data_path + '/MEG/spm/SPM_CTF_MEG_example_faces%d_3D.ds'
events = mne.find_events(raw, stim_channel='UPPT001')

raw = mne.io.read_raw_ctf(raw_fname % 1, preload=False)
picks = np.array([i for i,item in enumerate(raw.info['ch_names']) if 'O' in item]) #occipital channels
reject = {'mag': 4e-12}
epochs = mne.Epochs(raw, events=events, tmin=-0.1, baseline=(None, 0.0),
                    tmax=5, picks=picks, preload=True, reject=reject)

epochs.save('test-epo.fif.gz')
epochs_from_file =  mne.epochs.read_epochs('test-epo.fif.gz')
```

agramfort, 2017-11-07:
the problem is that MNE saves CTF files without compensation to reading and
writing gives you the same files. Now when you pick channels in CTF
you cannot redo the compensation on read as you miss channels and our code
breaks.

What we should do is to save files with compensated channels and print a
warning.

larsoner, 2017-11-20:
If people have picked channels out of the data that are necessary for compensation, I'm thinking we should `warn` and eliminate those `comps` from the measurement info.

agramfort, 2017-11-21:
agreed. If you picked channels there is no way we can save back
uncompensated data.

larsoner, 2018-02-27:
Closing as dup of #4241

closed 2018-02-27 by larsoner