mne-tools/mne-python #4682: Get MNE to work with Binder

choldgraf, 2017-10-19:
I think we're close with the `environment.yml` file, but I ran into an error when trying to build with Binder:

```
      File "/srv/conda/lib/python3.6/site-packages/vtk/vtkRenderingOpenGL2.py", line 9, in <module>
        from vtkRenderingOpenGL2Python import *
    ModuleNotFoundError: No module named 'vtkRenderingOpenGL2Python'
```

@larsoner thinks that we may need get Xvfb installed as well...which should be possible with `apt.txt` and `postBuild` files!

larsoner, 2017-10-19:
does `postBuild` get sourced or just run? If it's the former the `export DISPLAY=:99.0` that I need will work, if it's the latter it will not. If that's the case, is there some way to specify env vars when launching / setting up binder?

choldgraf, 2017-10-19:
@yuvipanda thoughts on this? do we suggest setting environment variables with `postBuild`?

larsoner, 2017-10-19:
@choldgraf one thing I don't get -- I want to use `mne-tools/mne-python.git:/environment.yml` for my requirements but have the `.ipynb` files from `mne-tools/mne-tools.github.io:/dev/_downloads/*` shown, how do I configure it to do that? I had this:
```
    'binder': {'org': 'mne-tools',
               'repo': 'mne-python',
               'url': 'https://beta.mybinder.org/',
               'branch': 'master',
               'dependencies': op.join('..', 'environment.yml'),
               },
```
But that will point the wrong place for the `.ipynb` files (as they live with our website repo/URL).

I the thing I'd need to do is copy the `environment.yml` from `mne-python.git` to `mne-tools.github.io`?

larsoner, 2017-10-19:
... and actually we need to get `Xvfb` running *before* it tries to install all of the requirements of `environment.yml` (one of the requirements needs access to a screen to build), is there a `preBuild` I can use?

choldgraf, 2017-10-23:
just confirmed that `apt.txt` is run *before* the language-specific builds, so if we apt-install `Xvfb` then the conda install should find it

choldgraf, 2017-10-23:
though if we *really* need to run shell commands before building anything, a `preBuild` file seems reasonable tho we'd need to discuss on the r2d repo

larsoner, 2017-10-24:
> though if we really need to run shell commands before building anything

Yes it needs a display active to `pip install`, so it needs to be apt->Xvfb->pip and I don't think it's possible currently.

choldgraf, 2017-10-24:
ohhh....hmm. I assume this is for mayavi? I think that will be tough to implement with build files in binder...we may need to roll a dockerfile in this case :-(

larsoner, 2019-08-21:
We have a separate repo for this and it works now

closed 2019-08-21 by larsoner