mne-tools/mne-python #4689: pns_idx referenced before assignment, raw.plot_psd()

jwe080, 2017-10-20:
Hi,

I'm reading in EGI mff data with an ECG lead.  When I try to raw.plot_psd() I get the following error

/mne-python/mne/io/egi/egimff.py in _read_segment_file(self, data, idx, fi, start, stop, cals, mult)
    571                     s_end = s_start + samples_read
    572 
--> 573                     data_view = data[n_data1_channels:, s_start:s_end]
    574                     _mult_cal_one(data_view, block_data[:n_pns_channels],
    575                                   pns_idx,

UnboundLocalError: local variable 'pns_idx' referenced before assignment

I can make the plot if after I:
del raw._raw_extras[0]['pns_names']

because that's the variable whose length is checked in that case, however this is likely not the best solution.

larsoner, 2017-10-20:
Is it easy for you share a small file that we can use to reproduce locally?

jwe080, 2017-10-23:
Attached is a smallish file.

raw = mne.io.read_raw_egi('_20170725_094007_mrr_qrs_obs.mff')
raw.plot_psd()

should reproduce the error (in '0.15.dev0').  The actual cap only has 128 channels so 129 is 'dead', still tracking down why the EGI file comes out that way...  There are 4 'stim' channels that show up too, which were originally events...

(what is the best way to debug these modules? I have a few other things I'm interested in adding but my print statement additions and module reload as a quick check to see what's happening don't seem to work... will try pdb next..) 

Thanks.

[test_data.zip](https://github.com/mne-tools/mne-python/files/1408795/test_data.zip)

larsoner, 2017-10-24:
> will try pdb next.

This is how I usually do it. You can use an IDE to set debug points, or if/when you hit an error do post-mortem debugging with `import pdb; pdb.pm()`

jwe080, 2017-10-24:
So I think the bug is on ~line 518 in egimff.py.  The pns data is read regardless of whether it is required ( or in chs_to_read => pns_chans is empty and pns_idx doesn't get assigned. )  Adding the check for len(pns_idx) (or just using that as the switch) on this line should fix things - at least it does for me in this instance.

if 'pns_names' in egi_info and len(pns_chans) > 0:

larsoner, 2017-10-24:
That fix sounds reasonable to me, can you make a PR to fix it?

PR #5078 (2018-04-02): MRG: Fix EGIMFF var
Closes #4689.

Seems overkill to add a test file for this one.
closed 2018-04-02 by larsoner