mne-tools/mne-python #4691: raw.plot error when adding annotation

cbrnr, 2017-10-23:
When I plot raw data, press 'a' and then 'Enter' I get the following error:

```
File "/usr/local/lib/python3.6/site-packages/mne/viz/utils.py", line 1880, in _annotations_closed
    params['ax'].selector.disconnect_events()
AttributeError: 'NoneType' object has no attribute 'disconnect_events'
```

This problem also existed in MNE 0.14.1.

agramfort, 2017-10-23:
with what data?

cbrnr, 2017-10-23:
Some data set we've recorded. I'm currently trying to replicate with a sample data set (but the download process from OSF is really slow).

cbrnr, 2017-10-24:
OK, this happens with any data.

agramfort, 2017-10-24:
how do I replicate?

cbrnr, 2017-10-24:
```
import mne
from mne.datasets import sample

data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_fname, preload=True)
raw.plot()
```

Then press 'a' and in the annotations window click on "Add label" - voilà!

larsoner, 2017-11-07:
I can't replicate in my environment. Maybe it's a `matplotlib` backend issue?

cbrnr, 2017-11-07:
Hm, did you test this on a Mac (with the default `osx` backend)?

larsoner, 2017-11-07:
No, Linux. On my OSX machine with the MacOSX backend I can't get the button presses to go to the window instead of the terminal (even after clicking in the window, etc.).

cbrnr, 2017-11-08:
I get the same error on Linux. Did you press "a" and then click on the "Add label" button?

cbrnr, 2017-11-08:
And this happens as far back as 0.14, which is before I made any commits to this module :relieved: - probably some update to Matplotlib introduced this error.

larsoner, 2017-11-08:
> Did you press "a" and then click on the "Add label" button?

Yes

Can't add a simple conditional `if params['ax'].selector is not None` to fix this?

cbrnr, 2017-11-08:
Strange, maybe you have a different Matplotlib or Qt or some other dependent lib version.

> Can't we add a simple conditional if params['ax'].selector is not None to fix this?

Yes, I'll do that (now that I'm pretty sure it's caused by a change in behavior of either Matplotlib or its deps).

PR #4739 (2017-11-09): [MRG] Fix AttributeError when closing the annotation window
Fix #4691
closed 2017-11-09 by agramfort