mne-tools/mne-python #4694: support dense head surfaces in `plot_alignment`

drammock, 2017-10-23:
There does not appear to be a way to specify whether to use the sparse or dense head in `mne.viz.plot_alignment`.  According to @larsoner this used to be possible with the now-deprecated `plot_trans` function.  Requesting an enhancement to allow this via `plot_alignment`.

The loading of the head surface is handled [here](https://github.com/mne-tools/mne-python/blob/master/mne/viz/_3d.py#L653-L659) so presumably that's where code to allow user-specified surface density would get plugged in.

agramfort, 2017-10-24:
+1

PR welcome

larsoner, 2017-10-24:
We could make things like `surfaces='sample-head-dense'` would work by looking for `.surf` and `.fif` extensions, and loading the surface or BEM surface, respectively. WDYT?

agramfort, 2017-10-24:
I would use head-dense.

cf doc:

*surfaces* : str | list

Surfaces to plot. Supported values: ‘head’, ‘outer_skin’, ‘outer_skull’,
‘inner_skull’, ‘brain’, ‘pial’, ‘white’, ‘inflated’. Defaults to (‘head’,).

larsoner, 2017-10-24:
That's doable but a departure from what is done now. The existing logic actually assumes 1) you've given something like the entire filename as it is in `$SUBJECTS_DIR/$SUBJECT/bem/`, and 2) you're loading a `.surf` file (or one of the surfaces in your already-loaded `bem`). So in cases where people had `sample-head-dense.fif`, it wouldn't load even if supplied `head-dense.fif` because it doesn't prepend the `sample-` at the beginning. We could add this prepending behavior (as well as the ability to also look for and load `.fif`) -- is that what you're thinking?

It looks like the existing code also doesn't check to see that all surfaces supplied were found (so `surfaces='foo'`) won't raise an error) so we should fix that while we're at it.

agramfort, 2017-10-24:
sample-head-dense.fif is a standard name. I think we can guess it.

+1 for the fix

PR #4705 (2017-10-25): MRG: Add plotting of dense head
Closes #4694.

@drammock can you see if this works for you?
closed 2017-10-25 by agramfort