mne-tools/mne-python #4695: MNE Coreg Exception in traits notification handler

ktavabi, 2017-10-24:
In python 2.7 conda venv `mne coreg` fails to load digitized head points due to 

> Exception occurred in traits notification handler.
> Please check the log file for details.
> Exception occurred in traits notification handler for object: <mne.gui._viewer.PointObject object at 0x7f79eafd7590>, trait: scene, old value: <mayavi.tools.mlab_scene_model.MlabSceneModel object at 0x7f79eb1b6170>, new value: <mayavi.tools.mlab_scene_model.MlabSceneModel object at 0x7f79eb46cf50>
> Traceback (most recent call last):
>   File "/home/ktavabi/miniconda3/envs/andrea-anders/lib/python2.7/site-packages/traits/trait_notifiers.py", line 519, in _dispatch_change_event
>     self.dispatch( handler, *args )
>   File "/home/ktavabi/miniconda3/envs/andrea-anders/lib/python2.7/site-packages/traits/trait_notifiers.py", line 482, in dispatch
>     handler( *args )
>   File "/home/ktavabi/Projects/mne-python/mne/gui/_viewer.py", line 225, in _plot_points
>     if not self.scene.running():
> AttributeError: 'MlabSceneModel' object has no attribute 'running'
> 

cc @larsoner 

[image]

closed 2017-10-29 by agramfort

dokato, 2017-12-07:
Hi @larsoner, is this fixed? I still have the same problem (recently reinstalled mne).

larsoner, 2017-12-07:
Yes it should be fixed, what is `mne.sys_info()`

dokato, 2017-12-07:
```
Platform:      Linux-3.10.0-327.3.1.el7.x86_64-x86_64-with-centos-7.2.1511-Core
Python:        2.7.13 |Anaconda, Inc.| (default, Sep 30 2017, 18:12:43)  [GCC 7.2.0]
Executable:    /home/user/anaconda3/envs/meg/bin/python
CPU:           x86_64: 12 cores
Memory:        188.7 GB

mne:           0.15
numpy:         1.10.2 {lapack=openblas, blas=openblas}
scipy:         0.19.1
matplotlib:    1.5.1

sklearn:       0.19.0
nibabel:       2.1.0
mayavi:        4.5.0
pycuda:        Not found
skcuda:        Not found
pandas:        0.20.3
```

larsoner, 2017-12-07:
You need to upgrade to `0.15.1` or `0.15.2` (latest release)

dokato, 2017-12-07:
Oh, sorry. I didn't notice that. It helped. Thanks!

PR #4701 (2017-10-29): MRG: Fix for old mayavi
Closes #4695.

@ktavabi can you see if this fixes things for you?