mne-tools/mne-python #4697: Misalignment of annotations after cropping raw

stfnrpplngr, 2017-10-24:
Hi,

I have an EEG dataset of one hour length covering several experiments. To mark the pauses between particular experiments as bad, I detect event codes marking their start and end point. After getting the corresponding time, I add it to the annotations variable and save it for further processing. The code is quite the same as in the [example](https://martinos.org/mne/dev/auto_tutorials/plot_artifacts_correction_rejection.html?highlight=blink%20annotation#marking-bad-raw-segments-with-annotations).

```
    raw = mne.io.read_raw_brainvision(raw_fname, montage=montage, preload=False)
    events = mne.find_events(raw, stim_channel='STI 014')

#%% Marking pauses as bad
    onset = np.array([events[idx,0] for idx in range(len(events[:, 2])) if events[idx, 2] > 80 and events[idx, 2] < 90])+1
    onset = np.insert(onset, 0, 0)
    end = np.array([events[idx,0] for idx in range(len(events[:, 2])) if events[idx, 2] > 70 and events[idx, 2] < 80])-1
    if events[0,2] < 70:
        end = np.concatenate([[events[0,0]-1], end, [len(raw)]])
    else:
        end = np.concatenate([end, [len(raw)]])

    duration = (end-onset)/raw.info['sfreq']
    onset = onset/raw.info['sfreq']
    description = ['bad']*len(onset)

    raw.annotations = mne.Annotations(onset, duration, description,
                                      orig_time=raw.info['meas_date'])

    raw.save(raw_fname+'.raw.fif.gz')
```

In a subsequent step, I load the EEG data, crop it and concatenate it. That results in a continous stream containing only the occurences of a certain experiment. 

```
    raw = mne.io.read_raw_fif(filename, preload=False)
    events = mne.find_events(raw, stim_channel='STI 014')

#%% cropping - assessing the beginning of particular experiments
    event_id = experiments['resting']['event_id']
    color = experiments['resting']['color']

# indices of the start and end triggercodes are detected, +1 and -1 to take a neighboring sample
    times_start = events[:, 0][np.where(events[:, 2] == event_id['start'])]-1
    times_end = events[:, 0][np.where(events[:, 2] == event_id['end'])]+1

    old_raw = raw.copy()
    new_raw = raw.copy().crop(raw.times[times_start[0]],
                              raw.times[times_end[0]])
    for idx in range(1, len(times_start)):
        mne.concatenate_raws([new_raw,
                              raw.copy().crop(raw.times[times_start[idx]],
                                              raw.times[times_end[idx]])])
# data is loaded, now the samples are loaded to memory
    raw = new_raw
    new_raw.close()
    raw.load_data()
    events = mne.find_events(raw, stim_channel='STI 014')
```

That is probably not the most elegant way of doing it.

The main problem is that after cropping the annotations are not where they are supposed to be. The following pictures show what I mean. 

[image]

[image]

The annotations appear now at a totally different position which does not even fit to the onset values in the annotations variable:

```
old_raw.annotations.onset
Out[52]: 
array([    0.        ,   165.36399841,   609.80700684,   684.23101807,
        1065.14794922,  1497.01794434,  1589.45605469,  1963.6159668 ,
        2370.41699219,  2510.62597656,  2877.16308594,  3300.84399414,
        3431.56201172,  3873.2409668 ,  3942.5090332 ])

old_raw.first_samp
Out[53]: 0

new_raw.annotations.onset
Out[54]: array([ 437.35700684,  511.78101807])

new_raw.first_samp
Out[55]: 172450
```

I am not sure were the mistake happens. Probably, I misunderstood how the `orig_time` parameter works.
What is your suggestion?

larsoner, 2017-10-24:
> The main problem is that after cropping the annotations are not where they are supposed to be. 

If you are able to 1) add annotations to a raw object, 2) have them be at the correct times, 3) crop the data, and 4) have them be at the wrong times (shifted relative to the data the originally covered) -- and it sounds like this is what you're describing -- then it seems like a bug.

Can you try to simplify this down to a minimal example with the `sample` data that just adds a single annotation, plots it, crops, plots again showing a shift?

agramfort, 2017-10-24:
i think it's related to first_samp that should be adjusted when using crop.

cbrnr, 2017-10-25:
Recently I also ran into problems with that pesky `first_samp` attribute that nobody seems to be aware of but MNE is still using sometimes. Time to start a new discussion to get rid of it?

agramfort, 2017-10-25:
the purpose of first_samp if is to have an absolute axis of time so if you
crop you know when it was in the full record.

makes sense?

cbrnr, 2017-10-25:
I know what the purpose is, but this only works if you crop from the beginning of the file. What if I crop in the middle or end? What if I concatenate several cropped files? This is something we routinely do and `first_samp` doesn't help.

agramfort, 2017-10-25:
[image]

crop updates first_samp cf image. Do you see a potential problem with this?
now if you concatenate data from different runs you're indeed screwed as you don't know anymore more the absolute time of recording.

cbrnr, 2017-10-25:
Plus it only works with .fif files. An option to get rid of first_samp would be nice (which would magically perform everything that makes a file start at time zero, including events).

agramfort, 2017-10-25:
just set first_samp to 0 for non fif files? why is this not enough?

cbrnr, 2017-10-25:
Because people don't know about this behavior.

Maybe we could add a note to `Raw.append` saying that if the individual `raw` objects have a `first_samp` > 0 everything gets mangled, therefore setting `first_samp=0` is recommended in such a case.

agramfort, 2017-10-25:
+1 for adding a big warning when merging raws that don't have first_samp ==
0

larsoner, 2017-10-25:
... or maybe more completely, that don't all have `first_samp == 0` or `first_samp[1] == last_samp[0]`, etc. As this should still work `concatenate_raws([raw.copy().crop(0, 1-1./sfreq), raw.copy().crop(1, None)])` and be the same as `raw`.

stfnrpplngr, 2017-10-25:
Did I misunderstand something? Setting `first_samp` to `0` does not work.

```
new_raw.first_samp
Out[32]: 172450

new_raw.first_samp = 0
Traceback (most recent call last):

  File "<ipython-input-33-d0c36293b1ef>", line 1, in <module>
    new_raw.first_samp = 0

AttributeError: can't set attribute
```

agramfort, 2017-10-25:
we would need a setter for this.

it should be done in the reader.

stfnrpplngr, 2017-10-26:
I tried putting a setter to the reader, but did not succeed, would need some help or at least more explanation to do that.

What also works:

```
raw.first_samp
Out[4]: 240

raw._first_samps[0] = 0

raw.first_samp
Out[6]: 0
```

stfnrpplngr, 2017-10-26:
Here is a minimal example using the sample data:
```
import mne
import numpy as np
from mne.datasets import sample

print(__doc__)

data_path = sample.data_path()

fname = data_path + '/MEG/sample/sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(fname, preload=False)

events = mne.find_events(raw, stim_channel='STI 014')

#%% Marking pauses as bad
onset = np.array([events[idx,0] for idx in range(len(events[:, 2])) if events[idx, 2] == 5])+1
end = np.array([events[idx,0] for idx in range(len(events[:, 2])) if events[idx, 2] == 32])-1
end = np.delete(end, -1)

duration = (end-onset)/raw.info['sfreq']
onset = onset/raw.info['sfreq']
description = ['bad']*len(onset)

raw.annotations = mne.Annotations(onset, duration, description,
                                  orig_time=raw.info['meas_date'])

#%% cropping - getting indices of where to crop
times_start = np.array([0, len(raw.times)*1/4, len(raw.times)*2/4, len(raw.times)*3/4])
times_end = np.array([len(raw.times)*1/4, len(raw.times)*2/4, len(raw.times)*3/4, len(raw.times)])-1

old_raw = raw.copy()
new_raw = raw.copy().crop(raw.times[times_start[0]],
                          raw.times[times_end[0]])

for idx in range(1, len(times_start)):
    cropped_raw = raw.copy().crop(raw.times[times_start[idx]],
                                  raw.times[times_end[idx]])
    mne.concatenate_raws([new_raw, cropped_raw])

old_raw.plot()
new_raw.plot()
```

First, I automatically set annotations according to event codes in `old_raw`. Then, I cut the `old_raw` data into four pieces which are subsequently concatenated. Intuitively, I would guess that the resulting `new_raw` should look the same as `old_raw`. That is not the case.

agramfort, 2017-10-29:
@stfnrpplngr have a look at https://github.com/mne-tools/mne-python/pull/4717

I think it fixes your problem.

PR #4717 (2018-04-06): MRG: FIX time offset bug in raw.crop for Annotations
closes #4697
closed 2018-04-06 by agramfort