mne-tools/mne-python #4699: raw.pick_channels modify the raw object

aweinstein, 2017-10-24:
`raw.pick_channels` modify the `raw` instance. The following code shows the behavior:

```
from __future__ import print_function
import mne
import numpy as np

fs = 1
t = [1,2,3]
x = [1,1,1]

ch_types = ['eeg', 'eeg']
ch_names = ['ch1', 'ch2']

data = np.atleast_2d(np.vstack((x, x)))
info = mne.create_info(ch_names=ch_names, sfreq=fs, ch_types=ch_types)
raw = mne.io.RawArray(data, info)

print('Before', raw.ch_names)
new_raw = raw.pick_channels(['ch1'])
print('After', raw.ch_names)
```
Output:
```
Before ['ch1', 'ch2']
After ['ch1']
```
I think this is either a bug or an undocumented behavior.

The change happens here https://github.com/mne-tools/mne-python/blob/master/mne/channels/channels.py#L760:

```
pick_info(self.info, idx, copy=False)
```

Using `copy=True` change the behavior, but that will change the behavior of `_pick_drop_channels`.

larsoner, 2017-10-24:
It is meant to do this by design (all `object.` methods should work inplace like this when possible), do you have time to make a PR to mention this inplace operation?

larsoner, 2017-10-24:
FYI to avoid such behavior you can do `raw_pick = raw.copy().pick_channels(...)`

closed 2017-10-24 by agramfort

aweinstein, 2017-10-24:
I'll do a PR by tomorrow.