mne-tools/mne-python #4709: unique channel names

jeremygrd, 2017-10-25:
I'm starting using MNE, and I'm using .cnt files. My problem is that when I try to import the file as a raw file using : mne.io.read_raw_cnt() I get message error saying that the 'channel names are not unique, found duplicate for: set([NULL]). Do you have any suggestions ? I am also using BrainVision file but with this format, this time there is no such error message. Thanks in advance for your suggestions.
here's the link of the file which is too heavy for github post
 https://drive.google.com/file/d/0B2VBJOUPv8tJWnV6aFVyRGhLM1k/view?usp=sharing

agramfort, 2017-10-29:
what version of mne are you using?

I tried with version 0.15 and I got:

```
In [8]: raw.ch_names
Out[8]:
['H1',
 'H2',
 'H3',
 'H4',
 'H5',
 'H6',
 'H7',
 'H8',
 'H9',
 'H10',
 'H11',
 'H12',
 'H13',
 'H14',
 'H15',
 'Hp1',
 'Hp2',
 'Hp3',
 'Hp4',
 'Hp5',
 'Hp6',
 'Hp7',
 'Hp8',
 'Hp9',
 'Hp10',
 'Hp11',
 'Hp12',
 'Hp13',
 'Hp14',
 'Hp15',
 'EOGR',
 'EOGL',
 'A1',
 'A2',
 'A3',
 'A4',
 'A5',
 'A6',
 'A7',
 'A8',
 'A9',
 'A10',
 'A11',
 'A12',
 'A13',
 'A14',
 'A15',
 'Ap1',
 'Ap2',
 'Ap3',
 'Ap4',
 'Ap5',
 'Ap6',
 'Ap7',
 'Ap8',
 'Ap9',
 'Ap10',
 'Ap11',
 'Ap12',
 'Ap13',
 'Ap14',
 'Ap15',
 'NULL-0',
 'NULL-1',
 'SA1',
 'SA2',
 'SA3',
 'SA4',
 'SA5',
 'SA6',
 'SA7',
 'SA8',
 'SA9',
 'SA10',
 'SA11',
 'SA12',
 'SA13',
 'SA14',
 'SA15',
 'SAp1',
 'SAp2',
 'SAp3',
 'SAp4',
 'SAp5',
 'SAp6',
 'SAp7',
 'SAp8',
 'SAp9',
 'SAp10',
 'SAp11',
 'SAp12',
 'SApp3',
 'SAp14',
 'SAp15',
 'NULL-2',
 'NULL-3',
 'Tp1',
 'Tp2',
 'Tp3',
 'Tp4',
 'Tp5',
 'Tp6',
 'Tp7',
 'Tp8',
 'Tp9',
 'Tp10',
 'Tp11',
 'Tp12',
 'Tp13',
 'Tp14',
 'Tp15',
 'T1',
 'T2',
 'T3',
 'T4',
 'T5',
 'T6',
 'T7',
 'T8',
 'T9',
 'T10',
 'NULL-4',
 'NULL-5',
 'NULL-6',
 'NULL-7',
 'NULL-8',
 'STI 014']
```

see numbers appended.

please close if you confirm it's working.

jeremygrd, 2017-10-29:
Thanks, it works, I did not use the latest version of MNE.

closed 2017-10-29 by jeremygrd