mne-tools/mne-python #4710: stim_channel="auto" in read_raw_edf renames stim channel

cbrnr, 2017-10-25:
The new default for `mne.io.read_raw_edf` in MNE 0.16 will be `stim_channel="auto"`. Using this argument, the stim channel is called `Status` instead of the usual `STI 014`. This causes `mne.find_events` to fail because by default it looks for a channel called `STI 014`. Is this intended?

cbrnr, 2017-10-25:
Not only that, but when I use `stim_channel="auto"` the resulting stim channel in the data contains negative values whereas it looks normal with the previous default of taking the last channel.

agramfort, 2017-10-25:
there is maybe a bug but you can can set a default name for your stim
channels with the set_config

larsoner, 2017-10-25:
> This causes mne.find_events to fail because by default it looks for a channel called STI 014. Is this intended?

Different systems sometime have different names. FWIW the default for the last number of years (10 maybe?) for Neuromag has even been STI101, so all newer systems use this. We usually use STI 014 when we synthesize a new events channel only. So this part is okay I think.

> when I use stim_channel="auto" the resulting stim channel in the data contains negative values whereas it looks normal with the previous default of taking the last channel.

This sounds like a bug

cbrnr, 2017-10-25:
Using the exact same file, the stim channel is called `STI 014` if `stim_channel=True` (the current default) and `Status` if `stim_channel="auto"` (the default in 0.16). This is also a bug. I'm currently trying to find a public EDF file with a stim channel to reproduce this behavior.

cbrnr, 2017-10-25:
The problem seems to occur in BDF files and not in EDF files. To reproduce, download the [BDF test files](https://www.biosemi.com/download/BDFtestfiles.zip) (unzip) and run:

```
print(mne.io.read_raw_edf("BDFtestfiles/Newtest17-256.bdf", preload=True, stim_channel=True).info["ch_names"][-1])
print(mne.io.read_raw_edf("BDFtestfiles/Newtest17-256.bdf", preload=True, stim_channel="auto").info["ch_names"][-1])
```

agramfort, 2017-10-25:
sorry no time to look.

cbrnr, 2017-11-07:
@larsoner do you have time to confirm?

agramfort, 2017-11-07:
now on current master I get :

```
~/work/src/mne-python/mne/io/edf/edf.py in _get_info(fname, stim_channel, annot, annotmap, eog, misc, exclude, preload)
    562     # These are the conditions under which a stim channel will be interpolated
    563     if stim_channel is not None and not (annot and annotmap) and \
--> 564             tal_channel is None and n_samps[stim_channel] != int(max_samp):
    565         warn('Interpolating stim channel. Events may jitter.')
    566     info._update_redundant()

ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
```

:(

larsoner, 2017-11-07:
I think we need somebody with EDF expertise/knowledge to tackle this one. @cbrnr do you know enough to have an idea of what the right thing would be to do in each of these situations? If so, maybe a table or bulleted list would be good to guide fixes and also add to the `Notes` docstring.

cbrnr, 2017-11-08:
No, I'm not familiar enough with the internal workings of EDF to do this quickly. I assume @jaeilepp is not available anymore, right? In the worst case, we could revert his changes regarding the stim channel.

larsoner, 2017-11-08:
> In the worst case, we could revert his changes regarding the stim channel.

This seems like a reasonable choice to me, but it's complicated by the changes already existing in 0.15 (and it might work properly for some users in `'auto'` mode, and we would cause them pain with removal). So it's probably better to fix it rather than remove it.

cbrnr, 2017-11-14:
Cf #4750

cbrnr, 2017-11-16:
The last channel in the test file I've linked above is the stim channel. Since the previous default was `-1`, this channel was correctly detected as a stim channel. Now the new default is `'auto'`, which results in the last channel being interpreted as an EEG channel (and hence the weird values).

What exactly does `stim_channel='auto'` do?

cbrnr, 2017-11-16:
To answer my question, maybe `stim_channel='auto'` should behave differently. Currently, it adds a stim channel if EDF annotations are present. If not, no stim channel is added. Maybe it should add the last channel in the latter case instead. Thoughts?

cbrnr, 2017-11-16:
It's a mess. I'm discovering weird behavior as I go. I thought that `stim_channel=None` wouldn't add a stim channel. Consider [this example EDF file](https://www.physionet.org/pn4/eegmmidb/S001/S001R04.edf). If I load it with `stim_channel=None`, an additional MISC channel is still appended, even though this file does not have such a stim channel (events are stored as EDF annotations).

larsoner, 2017-11-16:
It sounds like there are now several problematic files. If you have time, making a bulleted list (one per file) at the top of one of these issues that contains file link + existing behavior + what it should do would go a long way to setting up for a proper fix.

cbrnr, 2017-11-16:
#4760

cbrnr, 2017-11-18:
I think the backport label can be removed, because nothing is really broken (now that I understand the internals of EDF better), but only less ideal from a usability perspective.

agramfort, 2017-11-18:
cool. can you document this somewhere ? and maybe suggest API improvements?

cbrnr, 2017-11-18:
Yeah, it's all in #4760.

cbrnr, 2017-11-22:
@agramfort @larsoner slightly OT, when is 0.15.1 planned to be released? The viz bug in raw.plot is pretty serious and 0.15.1 fixes it. This issue is for 0.16 because everything (EDF-related stuff) is still working in 0.15.

agramfort, 2017-11-23:
just pushed 0.15.1 https://pypi.python.org/pypi/mne

cbrnr, 2017-11-23:
Thanks @agramfort! So far `pip` isn't seeing the new version, but I'm hoping this won't take too long.

PR #4888 (2018-02-11): Fix stim channel detection in BDF and EDF+
After GDF support was added, the stim channel in BDF files is not detected automatically anymore. The previous default used to be `stim_channel=-1` (the last channel), but this changed to `stim_channel='auto'`.

In EDF+ files, the TAL channel (containing annotations and therefore the stim channel) was assigned MISC - I changed this to STIM.

The whole thing is still a bit messy, mostly due to the mix of EDF/BDF/GDF. Maybe a future PR could try to disentangle these file formats better. However, my quick fix hopefully solves the most important issue, na [... cut]
closed 2018-02-11 by agramfort